margelo/react-native-vision-camera #1346: 🐛 Execution failed for task ':react-native-vision-camera:externalNativeBuildDebug'

SeongwoonHong, 2022-11-17:
### What were you trying to do?

I'm trying to run on Android simulator but it does not work. IOS is working fine.
```
"react-native": "0.66.5"
 "react-native-vision-camera": "^2.14.1"
```

I've tried 
```
delete my node_modules folder
yarn install
cd android
./gradlew clean
./gradlew assembleDebug
```
but still the same.

### What happened instead?

```
* What went wrong:
Execution failed for task ':react-native-vision-camera:externalNativeBuildDebug'.
> Build command failed.
  Error while executing process /Users/seong/Library/Android/sdk/cmake/3.10.2.4988404/bin/ninja with arguments {-C /Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-vision-camera/android/.cxx/cmake/debug/armeabi-v7a VisionCamera}
  ninja: Entering directory `/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-vision-camera/android/.cxx/cmake/debug/armeabi-v7a'
  [1/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o 
  /Users/seong/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=armv7-none-linux-androideabi21 --gcc-toolchain=/Users/seong/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/seong/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/sysroot  -DVisionCamera_EXPORTS -I../../../../build/fbjni-0.3.0-headers.jar -I../../../../build/third-party-ndk/boost -I../../../../build/third-party-ndk/double-conversion -I../../../../build/third-party-ndk/folly -I../../../../build/third-party-ndk/glog -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/React -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/React/Base -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/ReactAndroid/src/main/jni -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/ReactAndroid/src/main/java/com/facebook/react/turbomodule/core/jni -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/ReactCommon -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/ReactCommon/callinvoker -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/ReactCommon/jsi -I/Users/seong/Documents/usmobile/react-native-app/node_modules/hermes-engine/android/include -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-reanimated/Common/cpp/headers/AnimatedSensor -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-reanimated/Common/cpp/headers/Tools -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-reanimated/Common/cpp/headers/SpecTools -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-reanimated/Common/cpp/headers/SharedItems -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-reanimated/Common/cpp/headers/Registries -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-reanimated/Common/cpp/headers/LayoutAnimations -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-reanimated/Common/cpp/hidden_headers -I../../../../src/main/cpp -DFOLLY_NO_CONFIG=1 -DFOLLY_HAVE_CLOCK_GETTIME=1 -DFOLLY_HAVE_MEMRCHR=1 -DFOLLY_USE_LIBCPP=1 -DFOLLY_MOBILE=1 -DON_ANDROID -DONANDROID -DFOR_HERMES=true -DJS_RUNTIME_HERMES=1 -O0 -fno-limit-debug-info  -fPIC   -std=gnu++14 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o -MF CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o -c ../../../../src/main/cpp/VisionCameraScheduler.cpp
  In file included from ../../../../src/main/cpp/VisionCameraScheduler.cpp:5:
  ../../../../src/main/cpp/VisionCameraScheduler.h:7:10: fatal error: 'Scheduler.h' file not found
  #include "Scheduler.h"
           ^~~~~~~~~~~~~
  1 error generated.
  [2/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JHashMap.cpp.o
  [3/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JImageProxy.cpp.o
  [4/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessorPlugin.cpp.o
  [5/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o 
  /Users/seong/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=armv7-none-linux-androideabi21 --gcc-toolchain=/Users/seong/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/seong/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/sysroot  -DVisionCamera_EXPORTS -I../../../../build/fbjni-0.3.0-headers.jar -I../../../../build/third-party-ndk/boost -I../../../../build/third-party-ndk/double-conversion -I../../../../build/third-party-ndk/folly -I../../../../build/third-party-ndk/glog -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/React -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/React/Base -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/ReactAndroid/src/main/jni -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/ReactAndroid/src/main/java/com/facebook/react/turbomodule/core/jni -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/ReactCommon -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/ReactCommon/callinvoker -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/ReactCommon/jsi -I/Users/seong/Documents/usmobile/react-native-app/node_modules/hermes-engine/android/include -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-reanimated/Common/cpp/headers/AnimatedSensor -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-reanimated/Common/cpp/headers/Tools -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-reanimated/Common/cpp/headers/SpecTools -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-reanimated/Common/cpp/headers/SharedItems -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-reanimated/Common/cpp/headers/Registries -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-reanimated/Common/cpp/headers/LayoutAnimations -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native-reanimated/Common/cpp/hidden_headers -I../../../../src/main/cpp -DFOLLY_NO_CONFIG=1 -DFOLLY_HAVE_CLOCK_GETTIME=1 -DFOLLY_HAVE_MEMRCHR=1 -DFOLLY_USE_LIBCPP=1 -DFOLLY_MOBILE=1 -DON_ANDROID -DONANDROID -DFOR_HERMES=true -DJS_RUNTIME_HERMES=1 -O0 -fno-limit-debug-info  -fPIC   -std=gnu++14 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o -MF CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o -c ../../../../src/main/cpp/VisionCamera.cpp
  In file included from ../../../../src/main/cpp/VisionCamera.cpp:3:
  ../../../../src/main/cpp/FrameProcessorRuntimeManager.h:13:10: fatal error: 'RuntimeManager.h' file not found
  #include "RuntimeManager.h"
           ^~~~~~~~~~~~~~~~~~
  1 error generated.
  [6/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o 
  /Users/seong/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=armv7-none-linux-androideabi21 --gcc-toolchain=/Users/seong/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/seong/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/sysroot  -DVisionCamera_EXPORTS -I../../../../build/fbjni-0.3.0-headers.jar -I../../../../build/third-party-ndk/boost -I../../../../build/third-party-ndk/double-conversion -I../../../../build/third-party-ndk/folly -I../../../../build/third-party-ndk/glog -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/React -I/Users/seong/Documents/usmobile/react-native-app/node_modules/react-native/React/Base -I/Users/seon [... cut ...]
Android (any device)

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

stevebrowndotco, 2022-11-17:
Just started getting this as well. I noticed it was working previously for 2.14.0 and not 2.14.1 (which was resolved from ^2.14.0 for me recently) . Just testing and will report back

stevebrowndotco, 2022-11-17:
To fix this problem you need to downgrade and pin the version `react-native-reanimated` to precisely `2.10.0`.
This is **only** the case for situations where `react-native-vision-camera@2.14.x`

SeongwoonHong, 2022-11-17:
@stevebrowndotco It solved this issue. thank you 😄

closed 2022-11-17 by SeongwoonHong

ahmadalghalban, 2022-12-12:
> @stevebrowndotco It solved this issue. thank you 😄

Hello How to solve this issue ??