margelo/react-native-vision-camera #1347: ❓ How to disable flash light all the time?

SeongwoonHong, 2022-11-18:
### Question

I tried
```
torch="off"
torch={false}
```
but at the time of taking a picture, flash light is on. Is there a way to disable this at all?

### What I tried

torch="off"
torch={false}

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

vgorte, 2022-11-18:
Did you try the [take photo](https://mrousavy.com/react-native-vision-camera/docs/api/classes/Camera#takephoto) example, which showcases how to enable / disable flash?

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy