margelo/react-native-vision-camera #1348: Ability to zoom in QR code automatically.

shubhamdeol, 2022-11-21:
### What feature or enhancement are you suggesting?

If QR code is located at far distance. User should be able to zoom into QR code automatically. 

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

If this feature is not part of this library roadmap. Can you Please suggest ways to do this?
Is there any Frame Processor which can let me calculate distance of QR code from camera and based upon that I can set zoom levels

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

EKOzkan, 2022-12-01:
[Vision Camera Code Scanner Plugin](https://github.com/rodgomesc/vision-camera-code-scanner/) returns corner coordinates of the detected QR code, you can use them for digital zoom. If you want an optical zoom, I think the QR code must be close to the center of the screen for it. So for an optical zoom you can check the corner coordinates of the qr code and based on their position, you can zoom optically with [Vision Camera`s zoom function.](https://mrousavy.com/react-native-vision-camera/docs/guides/zooming/)

mrousavy, 2022-12-12:
yep - exactly what @EKOzkan said 👍 

You can only zoom in towards the center though, not move around the zoom if that makes sense.

closed 2022-12-12 by mrousavy

jasbirrajrana, 2023-02-15:
@EKOzkan @mrousavy can you please explain this lil bit more by some kind of example maybe?

bennyli519, 2023-04-27:
@EKOzkan  @mrousavy  can help to offer a demo for this case?  don't know how to achieve `zoom in QR code automatically`.

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀 

Try upgrading to V3 if you can, it has a much more stable yet flexible foundation.

You can just increase zoom if the QR code widht/height is too small

If you appreciate me dedicating my free time to improving VisionCamera and implementing features like the Code Scanner, please consider [**sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy) to show your support.