margelo/react-native-vision-camera #1352: 🐛 Support for  kotlin-android-extensions Gradle plugin version 1.6.20 and higher.

maddeye, 2022-11-23:
### What were you trying to do?

Run my demo application

### Reproduceable Code

```tsx
Use a blank react native init project with gradle 8
```

### What happened instead?

Build Process crashes.

### Relevant log output

```shell
FAILURE: Build failed with an exception.

* What went wrong:
The Android Gradle plugin supports only kotlin-android-extensions Gradle plugin version 1.6.20 and higher.
The following dependencies do not satisfy the required version:
project ':react-native-vision-camera' -> org.jetbrains.kotlin:kotlin-gradle-plugin:1.5.30
```

### Device

Android Simulator

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

finger-guns, 2022-12-06:
any updates on this one?

maddeye, 2022-12-06:
Not really. I found, that Gradle 7.3.3 and Gradle Plugin 7.2.2 is the sweetspot to get everything running and use the current react. You can set this in Android Studio -> Open Module Settings -> Project. Hope this helps!

shohelahamad, 2022-12-12:
any update on that ? I am having the same error.

bilal-korir, 2023-01-04:
> > any update on that ? I am having the same error.
> 
> I added `kotlinVersion = "1.7.0"` (under `buildToolsVersion`) to `android/build.gradle` and error disappeared.

Thank you @sleaper that solves the issue on my side.

betanyeli, 2023-01-28:
> > any update on that ? I am having the same error.
> 
> I added `kotlinVersion = "1.7.0"` (under `buildToolsVersion`) to `android/build.gradle` and error disappeared.

Gracias buen hombre 👨‍💻 ! ✨ funcionó perfecto

GerardoTovar, 2023-02-05:
> > any update on that ? I am having the same error.
> 
> I added `kotlinVersion = "1.7.0"` (under `buildToolsVersion`) to `android/build.gradle` and error disappeared.

Muchas Gracias bro. El problema se resolvió @sleaper

wmeints, 2023-02-09:
Adding `kotlinVersion =1.7.0` to my `android/gradle.properties` file in my project fixed it for me. Please note that I've done this to the

NilayYadav, 2023-02-12:
After addding kotlinVersion ="1.7.0 " to my android/gradle.properties file in my project gave me rror: EPERM: operation not permitted and Emitted 'error' event on NodeWatcher instance at: i am using very latest version of react native cli 

Does anyone know how can i fix that?

imran1992, 2023-02-14:
> Adding `kotlinVersion =1.7.0` to my `android/gradle.properties` file in my project fixed it for me.

By `android/gradle.properties` do you mean `android/build.gradle`

DevineDecrypter, 2023-03-20:
issue still persists on react-native@0.71.3

edritech93, 2023-04-20:
still issue on react-native 0.71.7

mkultr4, 2023-04-28:
still issue on react-native@0.71.7

waleedshkt, 2023-05-13:
Follow this instruction, and it's working on RN 0.71.8
> > any update on that ? I am having the same error.
> 
> I added `kotlinVersion = "1.7.0"` (under `buildToolsVersion`) to `android/build.gradle` and error disappeared.

jonxssc, 2023-05-16:
Bug is still there, i have a temporary fix for everyone who faces this error:
go to `react-native-vision-camera` in the `node-modules` folder. There go to `android/build.gradle` and search for this line `org.jetbrains.kotlin:kotlin-gradle-plugin`, change it to: `org.jetbrains.kotlin:kotlin-gradle-plugin:1.6.20` save, then run `cd android && ./gradlew clean` and then restart your App, works for me, i hope this bug gets fixed soon

Jhon-Idrovo, 2023-05-21:
This is stopping me from using this in production. Please fix it @mrousavy

sergioprz, 2023-05-22:
Has anyone gotten it to work with rn 0.70.8?

jonxssc, 2023-05-22:
> Has anyone gotten it to work with rn 0.70.8? 

Yes, try my answer two messages above

batgerelt, 2023-05-30:
> Bug is still there, i have a temporary fix for everyone who faces this error: go to `react-native-vision-camera` in the `node-modules` folder. There go to `android/build.gradle` and search for this [...]

Worked for me Thank you :-)

amanzrx4, 2023-05-31:
Saved my day @sleaper. Idk when will i be senior enough to figure these ugly errors.

mrousavy, 2023-06-01:
That's because RN upgrades (or other libs that use Kotlin) break the version. There's not a lot I can do about this apart from just updating Kotlin version everytime.

thegreatzeus, 2023-06-05:
@mrousavy Please upgrade the version because only you can update the npm package. I would have to manually make changes to the library folder in `node_modules` every time I want to build locally and if I want to use EAS build I can't even do that. 

Please push this change. Thank you!

bosz, 2023-06-09:
> ndroid/gradle.properties

Thanks bro, life saver.

Ali-Kheiri, 2023-06-21:
> > any update on that ? I am having the same error.
> 
> I added `kotlinVersion = "1.7.0"` (under `buildToolsVersion`) to `android/build.gradle` and error disappeared.

@sleaper
thanks!

mianaamirshehzad, 2023-07-05:
> > any update on that ? I am having the same error.
> 
> I added `kotlinVersion = "1.7.0"` (under `buildToolsVersion`) to `android/build.gradle` and error disappeared.

Thank you so much for giving the solution. It has reduced my efforts.

mianaamirshehzad, 2023-07-05:
> > > any update on that ? I am having the same error.
> > 
> > 
> > I added `kotlinVersion = "1.7.0"` (under `buildToolsVersion`) to `android/build.gradle` and error disappeared.
> 
> Thank you so much for giving the solution. It has reduced my efforts.

My previous error is eliminated but now my screen turns blank when I start the camera. Any fix for it.?

ajdnakljdkjasdnasdad, 2023-08-10:
> 

I tried the method you provided but my apk just crashes as soon as I open it. Is there a way to fix this?

istashak, 2023-08-11:
I've literally tried every suggested fix on this thread, and none of them have worked. I can't get past a gradle sync.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

ohheyitskartik, 2023-08-23:
> @mrousavy Please upgrade the version because only you can update the npm package. I would have to manually make changes to the library folder in `node_modules` every time I want to build locally and if I want to use EAS build I can't even do that.
> 
> Please push this change. Thank you!

use npm patch-package to patch it on your end with your changes so it applies changes on every npm install until the new version comes out

matiasgdev, 2023-08-23:
> > @mrousavy Please upgrade the version because only you can update the npm package. I would have to manually make changes to the library folder in `node_modules` every time I want to build locally and if I want to use EAS build I can't even do that.
> > Please push this change. Thank you!
> 
> use npm patch-package to patch it on your end with your changes so it applies changes on every npm install until the new version comes out
> 
> edit : i checked the files, it picks your kotlin version from your android/build.gradle but only if you have it defined there and the name matches `"kotlinVersion"` make sure you have it in your ext config and the name matches, i was using `"kotlin_version"` as kotlin version variable which caused the issue.

This is the true answer, thank you so much!

Hakimji-Mustafa, 2023-09-01:
Adding kotlinVersion =1.7.0 to my android/gradle.properties file in my project fixed it for me.

closed 2023-09-01 by mrousavy

akash510, 2023-09-25:
Hi, i already change the kotlin version to 1.7.0 FAILURE: Build failed with an exception. But, still showing same error. Please help. 
**Error**
* What went wrong:
The Android Gradle plugin supports only Kotlin Gradle plugin version 1.5.20 and higher.
The following dependencies do not satisfy the required version:
project ':wakelock' -> org.jetbrains.kotlin:kotlin-gradle-plugin:1.3.50

**my code:**
buildscript {
    ext.kotlin_version = '1.7.0'
    repositories {
        google()
        mavenCentral()

    }

    dependencies {
        classpath 'com.android.tools.build:gradle:7.3.0'
        classpath 'org.jetbrains.kotlin:kotlin-gradle-plugin:1.7.0'
        classpath 'com.google.gms:google-services:4.3.15'
    }

mahammad72, 2024-02-01:
> > any update on that ? I am having the same error.
> 
> I added `kotlinVersion = "1.7.0"` (under `buildToolsVersion`) to `android/build.gradle` and error disappeared.

Thanks man you  save my time