margelo/react-native-vision-camera #1353: 🐛 Android build error when using frame processor: 'folly/dynamic.h' file not found

obasille, 2022-11-25:
### What were you trying to do?

First, thank you for this great library!! :)

It was working fine until I updated to 2.15.2 (I was using 2.14.1 reanimated 2.10 with before that).
I'm now getting a build error (on Windows) when trying to use a frame processor.

It's the not the same error as reported in issue #1283

I'm using the following package version:
react-native-vision-camera: 2.15.2
react-native-reanimated: 2.12.0

The builds works on EAS though, so it might be a Windows specific issue?

### Reproduceable Code

```tsx
npx create-react-native-app myapp -t with-typescript
cd myapp
yarn expo prebuild
yarn expo install react-native-vision-camera
yarn expo install react-native-reanimated
yarn expo run:android
sh
```

### What happened instead?

Getting an Android compilation error when building on Windows.

### Relevant log output

```shell
> Task :react-native-vision-camera:buildCMakeDebug[arm64-v8a] FAILED
C/C++: ninja: Entering directory `C:\Users\Olivier\Documents\dev\systemic\myapp\node_modules\react-native-vision-camera\android\.cxx\Debug\2x3o1x4w\arm64-v8a'
C/C++: In file included from ../../../../src/main/cpp/JSIJNIConversion.cpp:16:
C/C++: C:/Users/Olivier/Documents/dev/systemic/myapp/node_modules/react-native/ReactAndroid/src/main/jni\react/jni/NativeMap.h:11:10: fatal error: 'folly/dynamic.h' file not found
C/C++: In file included from ../../../../src/main/cpp/VisionCamera.cpp:3:
C/C++: In file included from ../../../../src/main/cpp/FrameProcessorRuntimeManager.h:14:
C/C++: In file included from ../../../../src/main/cpp/reanimated-headers/AndroidScheduler.h:8:
C/C++: In file included from C:/Users/Olivier/Documents/dev/systemic/myapp/node_modules/react-native/ReactAndroid/src/main/jni\react/jni/CxxModuleWrapper.h:10:
C/C++: In file included from C:/Users/Olivier/Documents/dev/systemic/myapp/node_modules/react-native/ReactAndroid/src/main/jni\react/jni/CxxModuleWrapperBase.h:13:
C/C++: C:/Users/Olivier/Documents/dev/systemic/myapp/node_modules/react-native/ReactCommon\cxxreact/CxxModule.h:15:10: fatal error: 'folly/dynamic.h' file not found
C/C++: In file included from ../../../../src/main/cpp/FrameProcessorRuntimeManager.cpp:5:
C/C++: In file included from ../../../../src/main/cpp/FrameProcessorRuntimeManager.h:14:
C/C++: In file included from ../../../../src/main/cpp/reanimated-headers/AndroidScheduler.h:8:
C/C++: In file included from C:/Users/Olivier/Documents/dev/systemic/myapp/node_modules/react-native/ReactAndroid/src/main/jni\react/jni/CxxModuleWrapper.h:10:
C/C++: In file included from C:/Users/Olivier/Documents/dev/systemic/myapp/node_modules/react-native/ReactAndroid/src/main/jni\react/jni/CxxModuleWrapperBase.h:13:
C/C++: C:/Users/Olivier/Documents/dev/systemic/myapp/node_modules/react-native/ReactCommon\cxxreact/CxxModule.h:15:10: fatal error: 'folly/dynamic.h' file not found
```

### Device

Not relevant

### VisionCamera Version

2.15.2

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

FDiskas, 2022-11-29:
In my case only the production build was failing:
Helped this https://mrousavy.com/react-native-vision-camera/docs/guides/frame-processors#android

obasille, 2022-12-02:
Hi @FDiskas, thanks for your reply. The link you send mentions disabling frame processors, but I need to use them :)

obasille, 2022-12-02:
Thanks @soralit !
`react-native-reanimated` version `2.12.0` is the version I'm having issue with, along with `react-native-vision-camera` version `2.15.2` (see my description of the issue).
Are you building on Windows? (as the issue I'm having seems specific to that OS)

obasille, 2022-12-02:
Thanks, as described in the issue I'm starting a new project from scratch so it's not a matter of removing node_modules or reinstalling dependencies.
Are you sure you were having the exact same issue? (there has been other C++ compilation errors reported recently but not this one)

obasille, 2022-12-08:
I've tried creating a new app on Mac using the exact same commands than given in the bug description and it compiled successfully. Don't know if I was lucky though because I tried only once :)

obasille, 2022-12-08:
I'm having the same compilation error using React Native without Expo (on Windows 10):
```sh
npx react-native init myapp
cd myapp
yarn add react-native-vision-camera
yarn add react-native-reanimated
npx react-native run-android
```

> myapp/node_modules/react-native/ReactCommon\cxxreact/CxxModule.h:15:10: fatal error: 'folly/dynamic.h' file not found
>   #include <folly/dynamic.h>

@mrousavy do you have any hindsight about this issue? I'd be happy to help but I don't know where to start. I'm very familiar wuth C++ but I'm quite ignorant regarding how React Native setup Android builds (I've mostly used Visual Studio). Just let me know if you have a few pointers for what I should be looking for to try identify this compilation issue.

Note; soralit has been having this issue intermittently on Mac, but it's happening 100% of the times on Windows.

obasille, 2022-12-14:
@tgreco what's the OS you're using to build? (win/mac/linux)

tgreco, 2022-12-14:
> @tgreco what's the OS you're using to build? (win/mac/linux)

Mac OS Ventura 13.0.1

mat2718, 2022-12-16:
are there any updates on this?

realsteel85, 2022-12-22:
With "react-native-vision-camera": "^2.15.2", and  "react-native-reanimated": "^2.13.0", i have still the same error when building for a AAB file..

muhamad-rizki, 2022-12-26:
if you have set a custom path for `REACT_NATIVE_DOWNLOADS_DIR` on your env, make sure it clean or delete any existing downloaded tar.gz files, in my case i have multiple folly-*.tar.gz files so I think vision camera pick the wrong one

CristhianMotoche, 2022-12-29:
I also solved this by removing `REACT_NATIVE_DOWNLOADS_DIR`, which I agree doesn't seem to be reusing the downloaded dependencies, and then clearing my dependencies:

```
unset REACT_NATIVE_DOWNLOADS_DIR
cd android && ./gradlew clean
cd .. && rm -rf node_modules/
yarn
npx expo prebuild
npx expo run:android
```

tgreco, 2022-12-29:
Unfortunately I'm not using expo or setting `REACT_NATIVE_DOWNLOADS_DIR` :(

tastydev, 2023-02-06:
I am facing the same issue on prouction only build (creating a aab file): 

```
node_modules/react-native/ReactAndroid/src/main/jni/react/jni/NativeMap.h:11:10: fatal error: &apos;folly/dynamic.h&apos; file not found&#10;
```

OS:  macOS Ventura 13.1
reanimated: 2.13.0
vision-camera: 2.15.4

normal builds to the emulator are fine..
Do anyone know how this can be fixed?

Stringsaeed, 2023-04-04:
The issue occurred with me when I had multiple project using the root `reactNativeDownloads` dir and the versions of `folly` were different from each other

all I had to do is to clear the root directory and everything worked!

that was my case

emmanuelvicenteRabbitmx, 2023-07-03:
Delete node_module and then execute `react-native start --reset-cache`

pke, 2023-08-17:
all the fixes here only work once, the next build will fail again.

closed 2023-09-01 by mrousavy