margelo/react-native-vision-camera #1356: 🐛 When I turn on frameProcessor prop on, app crashes with SIGABRT

impodracer, 2022-11-26:
### What were you trying to do?

- At first, I was trying to use `vision-camera-code-scanner` so I followed instruction of that plugin. But app crashed.
- After then, I tried another plugin but app crashed with same error.
- So I made a dummy frameProcessor. It doesn't work either.
- Same code works perfect just without frameProcessor prop. (commented that line)

Package versions that I use are below

> react-native 0.68.5
> react-native-vision-camera 2.15.2
> react-native-reanimated 2.13.0
> vision-camera-code-scanner 0.2.0
> react-native-firebase 16.4.6
> etc...

### What happened instead?

The app crashes with SIGABRT fatal error.

### Relevant log output

```shell
11-27 01:30:12.672 19487 19907 F libc    : Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 19907 (pool-48-thread-), pid 19487 (wahan.bros.beta)
11-27 01:30:13.257 19915 19915 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
11-27 01:30:13.257 19915 19915 F DEBUG   : Build fingerprint: 'samsung/d2xks/d2x:12/SP1A.210812.016/N976NKSU2HVJ4:user/release-keys'
11-27 01:30:13.257 19915 19915 F DEBUG   : Revision: '24'
11-27 01:30:13.257 19915 19915 F DEBUG   : ABI: 'arm64'
11-27 01:30:13.257 19915 19915 F DEBUG   : Processor: '3'
11-27 01:30:13.257 19915 19915 F DEBUG   : Timestamp: 2022-11-27 01:30:12.875461280+0900
11-27 01:30:13.257 19915 19915 F DEBUG   : Process uptime: 117s
11-27 01:30:13.257 19915 19915 F DEBUG   : Cmdline: com.sampleapp
11-27 01:30:13.257 19915 19915 F DEBUG   : pid: 19487, tid: 19907, name: pool-48-thread-  >>> com.sampleapp <<<
11-27 01:30:13.257 19915 19915 F DEBUG   : uid: 10562
11-27 01:30:13.257 19915 19915 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
11-27 01:30:13.258 19915 19915 F DEBUG   :     x0  0000000000000000  x1  0000000000004dc3  x2  0000000000000006  x3  0000006e9c93c2a0
11-27 01:30:13.258 19915 19915 F DEBUG   :     x4  00000000ebad808a  x5  00000000ebad808a  x6  00000000ebad808a  x7  00000000ebad808b
11-27 01:30:13.258 19915 19915 F DEBUG   :     x8  00000000000000f0  x9  00000072de06e290  x10 ffffff00fffffbdf  x11 0000000000000001
11-27 01:30:13.258 19915 19915 F DEBUG   :     x12 0000006f52d797e4  x13 0000000000000045  x14 0000000000000008  x15 0000000000000008
11-27 01:30:13.258 19915 19915 F DEBUG   :     x16 00000072de10f058  x17 00000072de0ebe00  x18 0000006e9c356000  x19 00000000000000ac
11-27 01:30:13.258 19915 19915 F DEBUG   :     x20 0000000000004c1f  x21 00000000000000b2  x22 0000000000004dc3  x23 00000000ffffffff
11-27 01:30:13.258 19915 19915 F DEBUG   :     x24 0000000000000001  x25 000000702a032e80  x26 0000000000000001  x27 00000071b9cba148
11-27 01:30:13.258 19915 19915 F DEBUG   :     x28 0000000000000001  x29 0000006e9c93c320
11-27 01:30:13.258 19915 19915 F DEBUG   :     lr  00000072de09d394  sp  0000006e9c93c280  pc  00000072de09d3c4  pst 0000000000001000
11-27 01:30:13.258 19915 19915 F DEBUG   : backtrace:
11-27 01:30:13.258 19915 19915 F DEBUG   :       #00 pc 00000000000513c4  /apex/com.android.runtime/lib64/bionic/libc.so (abort+180) (BuildId: 88b933fc529619f5e58bc6d5510017ee)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #01 pc 00000000000af7e4  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libc++_shared.so (BuildId: ece72a2ebc3774a1be9fd21271258acd3bcdfaa7)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #02 pc 00000000000aedc8  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libc++_shared.so (BuildId: ece72a2ebc3774a1be9fd21271258acd3bcdfaa7)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #03 pc 00000000000aebf8  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libc++_shared.so (__gxx_personality_v0+200) (BuildId: ece72a2ebc3774a1be9fd21271258acd3bcdfaa7)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #04 pc 000000000022bd3c  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libreanimated.so (BuildId: 3e7719d432cdb9c68067bc821ad5b7ee95fd524d)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #05 pc 000000000022bdf8  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libreanimated.so (BuildId: 3e7719d432cdb9c68067bc821ad5b7ee95fd524d)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #06 pc 00000000001ae798  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libreanimated.so (BuildId: 3e7719d432cdb9c68067bc821ad5b7ee95fd524d)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #07 pc 00000000001ae478  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libreanimated.so (BuildId: 3e7719d432cdb9c68067bc821ad5b7ee95fd524d)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #08 pc 00000000001ae3c4  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libreanimated.so (BuildId: 3e7719d432cdb9c68067bc821ad5b7ee95fd524d)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #09 pc 00000000001ae33c  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libreanimated.so (BuildId: 3e7719d432cdb9c68067bc821ad5b7ee95fd524d)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #10 pc 00000000001ad248  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libreanimated.so (BuildId: 3e7719d432cdb9c68067bc821ad5b7ee95fd524d)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #11 pc 0000000000031bac  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libhermes.so (BuildId: 4fbed506ba7b5e91b09f3a3ace7967fb54b2332c)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #12 pc 000000000003a858  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libhermes.so (BuildId: 4fbed506ba7b5e91b09f3a3ace7967fb54b2332c)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #13 pc 00000000000293a8  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libhermes.so (BuildId: 4fbed506ba7b5e91b09f3a3ace7967fb54b2332c)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #14 pc 00000000000d2034  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libVisionCamera.so (facebook::jsi::Function::callWithThis(facebook::jsi::Runtime&, facebook::jsi::Object const&, facebook::jsi::Value const*, unsigned long) const+108) (BuildId: 0aabef2bc6130caf07fd2af89231a6f7147c2616)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #15 pc 00000000000d1f64  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libVisionCamera.so (facebook::jsi::Function::callWithThis(facebook::jsi::Runtime&, facebook::jsi::Object const&, std::initializer_list<facebook::jsi::Value>) const+124) (BuildId: 0aabef2bc6130caf07fd2af89231a6f7147c2616)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #16 pc 00000000000d1ad0  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libVisionCamera.so (facebook::jsi::Value facebook::jsi::Function::callWithThis<facebook::jsi::Object>(facebook::jsi::Runtime&, facebook::jsi::Object const&, facebook::jsi::Object&&) const+148) (BuildId: 0aabef2bc6130caf07fd2af89231a6f7147c2616)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #17 pc 00000000000c0304  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libVisionCamera.so (BuildId: 0aabef2bc6130caf07fd2af89231a6f7147c2616)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #18 pc 00000000000c0234  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libVisionCamera.so (BuildId: 0aabef2bc6130caf07fd2af89231a6f7147c2616)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #19 pc 00000000000c01c0  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libVisionCamera.so (BuildId: 0aabef2bc6130caf07fd2af89231a6f7147c2616)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #20 pc 00000000000c0180  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libVisionCamera.so (BuildId: 0aabef2bc6130caf07fd2af89231a6f7147c2616)
11-27 01:30:13.258 19915 19915 F DEBUG   :       #21 pc 00000000000bf208  /data/app/~~aBG0AbCbMiWiwmXZn0vZwg==/com.sampleapp-SFMXKZQYkfg4-VStaM38yQ==/base.apk!libVisionCamera.so (BuildId: 0aabef2bc6130caf07fd2af89231a6f7147c2616)
[... cut ...]
Samsung Galaxy Note 10+

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mbpictures, 2023-01-27:
Any fixes or workarounds?

franzapiola, 2023-01-27:
I'm having the same issue, can't use frameProcessor prop

react-native 0.64.2
react-native-reanimated 2.2.0
react-native-vision-camera 2.13.0

explodingcamera, 2023-01-28:
For me, after a bunch of recompiling and re-generating the android/ios folders, the problem just went away. I'm also importing react-native-reanimated at the top of the file I'm using the camera in, but wasn't able to reproduce the issue consistently anymore after it was fixed.

CaioQuirinoMedeiros, 2023-02-01:
@explodingcamera can you please tell us the versions of react-native, react-native-reanimated and react-native-vision-camera that you are on?

explodingcamera, 2023-02-01:
> @explodingcamera can you please tell us the versions of react-native, react-native-reanimated and react-native-vision-camera that you are on?

```
"react-native": "0.70.5",
"react-native-reanimated": "~2.12.0",
"react-native-vision-camera": "^2.15.2",
"vision-camera-image-labeler": "^0.1.6",
```

Together with an expo 47 App that uses expo prebuild, so I compile all the code myself just like in vanilla react-native.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy