margelo/react-native-vision-camera #1359: When install react-native-vision-camera Build Error❓

mehmettalhairmak, 2022-11-27:
### Question

Hello,

I trying build my project after installed react-native-vision-camera but returning some errors

React Native version : 0.66.4
Gradle version : 4.2.2

build.gradle file ;

[image]

errors ;

[image]

### VisionCamera Version

^2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

IAmCoder, 2022-11-29:
Try changing you `compileSdkVersion` from 30 to 31.

mehmettalhairmak, 2022-12-02:
> Try changing you `compileSdkVersion` from 30 to 31.

Fixed. Thanks :)

closed 2022-12-02 by mehmettalhairmak