margelo/react-native-vision-camera #1360: iOS app with frameProcessor crashes after reload 🐛

sommcz, 2022-11-28:
### What were you trying to do?

When vision camera with any frameprocessor (eg. empty) is used in iOS app, it crashes after reload (both debug and release build). Reload can be triggered by pressing "r" in metro console or with react-native-restart Restart method in regular release (my case for production).

I tried different RN (hermes) versions and different reanimated versions, but behaviour is still the same.

### Reproduceable Code

```tsx
I prepared simple app based on latest RN (0.70.6) template and tested same behaviour.

https://github.com/sommcz/vision-camera-crash-test1

When I comment out frameProcessor prop, restart works as expected.
```

### What happened instead?

Exception happen in debug mode. For non-debug, app just stops (crash).

[image]

### Relevant log output

```shell
Exception: com.facebook.react.JavaScript (15): EXC_BAD_ACCESS
Line: HostFunctionHandler.h:11

Build with Xcode version 14
```

### Device

iPhone 12 mini

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Noah-Lc, 2022-11-30:
By the way, it's working fine on Xcode 13, but not on Xcode 14. I was following this, but this push didn't solve the problem.
[🐛 Vision Camera crashes when built with Xcode 14](https://github.com/mrousavy/react-native-vision-camera/issues/1244)

sommcz, 2022-11-30:
I have this problem in my older version built with Xcode 13 (RN 0.67 and some older version of vision-camera and reanimated). I can't test Xcode13 + latest libraries versions. Can anyone confirm that combinaton?

Noah-Lc, 2022-11-30:
`"react-native": "0.70.5",
"react-native-reanimated": "2.12.0",
"react-native-vision-camera": "^2.15.2",
"vision-camera-code-scanner": "^0.2.0"`
works on Xcode 13, and crash on Xcode 14

mrousavy, 2022-12-06:
Huh, that's interesting. Wonder what changed in Xcode 14...

I am working on VisionCamera V3, which will use a new Worklet Runtime to replace Reanimated's. This will fix this issue (and probably a ton of other issues related to FPs) and improve performance - but I need companies willing to help fund the development of this project - here's the full proposal for anyone interested: https://gist.github.com/mrousavy/1c8ac7559b88bf0712edd628b27dffea

DM me on Twitter / Email me if you're interested

TacticCoder, 2023-01-10:
Can confirm that this problem is not Xcode version related, at least on my side.
Test with Xcode 13.4
Test with Xcode 14.1
The same result, after the bundle reloads on IOS I get a crash.

The crash has led me to the  FrameProcessorRuntimeManager.

JYBESiU, 2023-01-21:
Same with Android

abejfehr, 2023-02-07:
This bug makes this library unusable for the purposes of scanning a QR code of the URL of a JS bundle to swap to

rocket13011, 2023-03-08:
We have the same problem, we have to reload the app. Is it possible to stop the frameprocessor completely? 

Because if I restart my app without going to my page with the active camera + frameprocessor + qrcode I have absolutely no problem restarting.
But when I go on it and leave the stack and restart even 2 minutes after the app crashes.

hungnd6282, 2023-07-18:
how to resolved?

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy