margelo/react-native-vision-camera #1365: 🐛 Can not save video > 1minute on android

nhuthuynh195, 2022-12-01:
### What were you trying to do?

When record video > 1 minute on android not save to device

### What happened instead?

When record video > 1 minute on android not save to device, when I use RNFS try read path will re return error file not found.

### Device

OnePlus 8T

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

nhuthuynh195, 2022-12-04:
Anyone can resolve issue?

mrousavy, 2022-12-06:
Hey! Oof, this looks like a native CameraX issue. This is Google's fault. Can you create an issue in their issue tracker?

mrousavy, 2022-12-06:
Actually wait - there is no error message here. This just says that the Camera has been destroyed (e.g. when you move to a new screen).

Can you show the full log output (scroll further up), maybe there's something interesting there?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy