margelo/react-native-vision-camera #1368: 🐛 Example app instantly crashes.

NcMasters, 2022-12-01:
### What were you trying to do?

Open the example app.

### What happened instead?

The app crashes immediately.

### Device

Samsung a52s

### VisionCamera Version

Latest Android example app

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

NcMasters, 2022-12-04:
It's really annoying

mrousavy, 2022-12-06:
Hey - you didn't put any logs here. I cannot help you. The issue template even has a field for "Relevant log output", and you left it blank.

The log output I'm looking for is the adb logcat output.

closed 2022-12-06 by mrousavy

NcMasters, 2022-12-06:
> Hey - you didn't put any logs here. I cannot help you. The issue template even has a field for "Relevant log output", and you left it blank.
> 
> The log output I'm looking for is the adb logcat output.

Im using a real phone. How do i get adb logcat output?

NcMasters, 2022-12-07:
> > Failed to load Reanimated/VisionCamera C++ library. Frame Processors are disabled!
> 
> hm 🤔 
> 
> It also can't find either JSC nor Hermes...

I am new to this, so maybe i didn't install everything?