margelo/react-native-vision-camera #1369: 🐛  error: package com.mrousavy.camera.frameprocessor does not exist

L-U-C-K-Y, 2022-12-01:
### What were you trying to do?

I all, I am trying for days to bundle my app, but I am always getting the following error.

I also tried different node versions, different build settings, but always end with the same error message...

### Reproduceable Code

```tsx
-
"react-native-vision-camera": "2.13.0",
"vision-camera-code-scanner": "^0.2.0",
```

### What happened instead?

```
error: package com.mrousavy.camera.frameprocessor does not exist
```

### Relevant log output

```shell
/.../<app>/node_modules/vision-camera-code-scanner/android/src/main/java/com/visioncameracodescanner/VisionCameraCodeScannerPluginPackage.java:9: error: package com.mrousavy.camera.frameprocessor does not exist
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;
                                         ^
```

### Device

Mabook Pro 2019, M1 Max

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-12-06:
Hey! Did you enable Frame Processors using the build flag (see docs)?

Also, did you maybe change your Kotlin versions? make sure you're using the ones that VisionCamera is using

tgreco, 2022-12-06:
I ran into the same issue. Didn't happen ever and then suddenly started happening and never went away.

OstinPower41016, 2023-04-27:
there was the same mistake, this helped me [click](https://github.com/mrousavy/react-native-vision-camera/issues/1039#issuecomment-1198697939)

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

DanielWoldehana, 2024-07-22:
@mrousavy I am on Android 14 sdk 34 and I get this error:

/node_modules/vision-camera-code-scanner/android/src/main/java/com/visioncameracodescanner/VisionCameraCodeScannerPluginPackage.java:9: error: package com.mrousavy.camera.frameprocessor does not exist

basically I think it is saying this import does not exist:
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin
Is there any solution to this, and Thank you in advance.

IgorDePaula, 2024-12-02:
He created this error on purpose to monetize his component. This component is no longer open source. Consider using another module.