margelo/react-native-vision-camera #1370: ❓ No controllers

androidealp, 2022-12-02:
### Question

my code is simple, but I don't see the click controls, only the open camera.
why this not working?

### What I tried

this works to show the camera, but it doesn't show the button controllers.

`    if(device == null) return (<Text>Carregando....</Text>)
    return (<Camera
                        style={StyleSheet.absoluteFill}
                        device={device}
                        isActive={true}
                    />)
`

### VisionCamera Version

^2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

EKOzkan, 2022-12-02:
Vision Camera library is designed for multiple use cases, so the "Camera" component returns a simple camera page with no controllers, to give project based customizability. If you want to add the button controllers, you can take a look at the [example project](https://github.com/mrousavy/react-native-vision-camera/tree/main/example).

androidealp, 2022-12-02:
> Vision Camera library is designed for multiple use cases, so the "Camera" component returns a simple camera page with no controllers, to give project based customizability. If you want to add the [...]

Thanks @EKOzkan

mrousavy, 2022-12-06:
Thanks for the answer @EKOzkan, that's correct :)

closed 2022-12-06 by mrousavy