margelo/react-native-vision-camera #1372: App crashes when set video={true} on my android phone. urgent. please help anyone

Robogram, 2022-12-02:
### What were you trying to do?

I am trying to record videos on my app on android phones

### Reproduceable Code

```tsx
<Camera
                ref={camera}
                style={{ height: '100%', width: '100%' }}
                device={type == 'front' ? devices.front : devices.back}
                isActive 
                video={true}
                photo audio
              />
```

### What happened instead?

App crashes with 'app keeps stopping'

### Device

galaxy a03 core, android 11

### VisionCamera Version

2.15.2

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

EKOzkan, 2022-12-03:
Did you get the both, camera and microphone permissions?

Robogram, 2022-12-03:
const getCameraPermission = async action => {
    setPermissions({ ...permissions, camera: false, audio: false })

    const camPermission = await Camera.getCameraPermissionStatus()
    const microphonePermission = await Camera.getMicrophonePermissionStatus()

    let camera = false, audio = false

    if (camPermission != "authorized") {
      const requestCam = await Camera.requestCameraPermission()

      camera = requestCam == "authorized"
    } else {
      camera = true
    }

    if (microphonePermission != "authorized") {
      const requestAudio = await Camera.requestMicrophonePermission()

      audio = requestAudio == "authorized"
    } else {
      audio = true
    }

    setPermissions({ ...permissions, camera, audio, action })
  }

(devices.front && permissions.camera == true && permissions.audio == true) && (
                          <Camera
                            style={{ height: '100%', width: '100%' }}
                            ref={camera}
                            device={createPostbox.cameraType == 'front' ? devices.front : devices.back}
                            isActive={true}
                            photo={true}
                            video={true}
                            audio={true}
                          />
                        )

Robogram, 2022-12-07:
I opened two terminal. One running 'npx react-native log-android' and one running the application. I don't see any logs. The only thing I see is the app closing whenever camera is about to open if I set video=true.

mrousavy, 2022-12-07:
Run `adb logcat` in Terminal (or run the app through Android Studio)

Robogram, 2022-12-10:
Hi, this is what my logcat output:

FATAL EXCEPTION: main
                                                                                                    Process: com.robogram.testapp, PID: 2849
                                                                                                    java.lang.IllegalStateException: Surface was requested when the Recorder had encountered error androidx.camera.video.internal.ResourceCreationException: androidx.camera.video.internal.encoder.InvalidConfigException: Encoder cannot created: null
                                                                                                    	at androidx.camera.video.Recorder.onSurfaceRequested(Recorder.java:406)
                                                                                                    	at androidx.camera.video.VideoCapture.createPipeline(VideoCapture.java:416)
                                                                                                    	at androidx.camera.video.VideoCapture.resetPipeline(VideoCapture.java:482)
                                                                                                    	at androidx.camera.video.VideoCapture.lambda$createPipeline$0$androidx-camera-video-VideoCapture(VideoCapture.java:433)
                                                                                                    	at androidx.camera.video.VideoCapture$$ExternalSyntheticLambda0.onError(Unknown Source:10)
                                                                                                    	at androidx.camera.camera2.internal.Camera2CameraImpl.lambda$postSurfaceClosedError$14(Camera2CameraImpl.java:1175)
                                                                                                    	at androidx.camera.camera2.internal.Camera2CameraImpl$$ExternalSyntheticLambda7.run(Unknown Source:4)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:938)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loop(Looper.java:223)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:7815)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:593)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1094)

Robogram, 2022-12-10:
Hi, i followed this link and i fixed my problem.

I had to add these lines in "android/app/build.gradles" inside the "dependencies" script.

implementation "androidx.swiperefreshlayout:swiperefreshlayout:1.0.0"
    implementation "androidx.camera:camera-core:1.2.0-beta02"
    implementation "androidx.camera:camera-camera2:1.2.0-beta02"
    implementation "androidx.camera:camera-lifecycle:1.2.0-beta02"
    implementation "androidx.camera:camera-video:1.2.0-beta02"
    implementation "androidx.camera:camera-view:1.2.0-beta02"
    implementation "androidx.camera:camera-extensions:1.2.0-beta02"

I thank you for your hard work on creating the 'react-native-vision-camera' package but I strongly suggest you update the package to include these updates so that others won't have the same problem.

Cheers:)

xeroxoid, 2023-01-24:
Possible duplicate of #907 (solution inside)