margelo/react-native-vision-camera #1374: Execution failed for task ':react-native-vision-camera:extractJNIFiles'.

ahmadalghalban, 2022-12-05:
### What were you trying to do?

when yarn android  
return this error 
Execution failed for task ':react-native-vision-camera:extractJNIFiles'. > Cannot expand ZIP
Execution failed for task ':react-native-vision-camera:extractJNIFiles'.

### What happened instead?

Execution failed for task ':react-native-vision-camera:extractJNIFiles'. 
and Don't build in android
Execution failed for task ':react-native-vision-camera:extractJNIFiles'.
> Cannot expand ZIP '/Users/shopyardshopyard/Documents/GitHub/Dare-Me-App/node_modules/react-native-reanimated/android/react-native-reanimated-65-hermes.aar' as it does not exist.

### Relevant log output

```shell
Execution failed for task ':react-native-vision-camera:extractJNIFiles'. 
and Don't build in android
Execution failed for task ':react-native-vision-camera:extractJNIFiles'.
> Cannot expand ZIP '/Users/shopyardshopyard/Documents/GitHub/Dare-Me-App/node_modules/react-native-reanimated/android/react-native-reanimated-65-hermes.aar' as it does not exist.
```

### Device

android

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

carlgrimsborn, 2023-02-09:
getting the same issue

hoseinfo, 2023-02-12:
same here,
any update?

EddyDunda, 2023-04-18:
Any updates? I got the same

aqigif, 2023-05-04:
any update? i got the same after upgrading RN from rn 0.71.7 to 0.72.0-rc.0

chakravarthi-bb, 2023-06-30:
Reanimated should build and release 72 aar so it gets fixed https://github.com/software-mansion/react-native-reanimated/issues/4522#issuecomment-1598936003

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy