margelo/react-native-vision-camera #1375: ❓ [Andoird][Orientation] is always as landscape.

Emilien1982, 2022-12-06:
### Question

Hi, 
The frame recieved by the frameProcessor is always in landscape mode (width > height) under my Android.
I expected the same behavior than iOS: width and height switches on every orientation change.

Is that normal for Android? or it's likely a bug ?

my Android: Samsung S21, under Android 13

### What I tried

console.log() of `frame` or `frame.toString()` in the frameProcessor when in dev mode.
With Android it always show width > height.

### VisionCamera Version

 2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-12-06:
Hm, this might be a bug. Here is where I configure the orientation: https://github.com/mrousavy/react-native-vision-camera/blob/231838772cddc0a837e7f3cf8bf4c077d3cc8fab/android/src/main/java/com/mrousavy/camera/CameraView.kt#L266 - can you confirm that this is being called? (e.g. by setting a breakpoint or putting a Log.d there)

mrousavy, 2022-12-06:
What happens if you pass `orientation="portrait"` or `orientation="landscape"` to the `<Camera>`?

Emilien1982, 2022-12-06:
Hih Marc,
Thanks to take time to reply.

I tried orientation with "portrait", "landscapeLeft" then "landscapeRight". Works fine on iOS but no effect on Android : no matter the orientation props and the device real orientation, the frame height and width stay the same.

mrousavy, 2022-12-07:
I'll take a look when I do [VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376)

Reizar, 2022-12-19:
Hi @mrousavy ,

I've been trying to look into this too as its affecting my app. 

Changing the orientation did nothing, I also tried reversing the width and heights on the camera format. But that didn't do anything either.

I added some log statements to the updateOrientation function, and it does get called. I also confirmed that the inputOrientation and outputOrientation inside that function were both the same.

zhuweideng, 2023-01-16:
@mrousavy i meet same issue.orientation with "portrait", "landscapeLeft" then "landscapeRight" is not effect on Android

zhuweideng, 2023-01-29:
@Emilien1982 i solve this issue by do this in 'AndroidManifest.xml'
`
<application
android:hardwareAccelerated="true"
...
>
`

Emilien1982, 2023-02-01:
> 
I tried it but it doesn't have any effect. On Android, frame.height and frame.width are always identical, no matter the device is hold as a portrait or a landscape.

zhuweideng, 2023-02-02:
> 

i have no idea about your issue

zhuweideng, 2023-02-02:
> @zhuweideng I tried it but it doesn't have any effect. On Android, frame.height and frame.width are always identical, no matter the device is hold as a portrait or a landscape.

i have no idea about your issue

mrousavy, 2023-09-30:
Orientation is quite complex - I'll try to tackle this in a future version if I get enough sponsors to support me. Why it's always landscape; Camera sensors are in 90deg orientation.

closed 2023-09-30 by mrousavy