margelo/react-native-vision-camera #1377: Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.🐛

AhmadAl-Ghalban, 2022-12-06:
### What were you trying to do?

when run app return this error 
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.

### What happened instead?

> Task :react-native-vision-camera:compileDebugKotlin FAILED

### Relevant log output

```shell
See https://docs.gradle.org/7.2/userguide/command_line_interface.html#sec:command_line_warnings
563 actionable tasks: 547 executed, 16 up-to-date
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\@react-native-async-storage\async-storage\android\src\main\java\com\reactnativecommunity\asyncstorage\AsyncStorageModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\@react-native-async-storage\async-storage\android\src\main\java\com\reactnativecommunity\asyncstorage\AsyncStoragePackage.java uses unchecked or unsafe operations.        
Note: Recompile with -Xlint:unchecked for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\react-native-background-timer\android\src\main\java\com\ocetnik\timer\BackgroundTimerModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\@react-native-community\checkbox\android\src\main\java\com\reactnativecommunity\checkbox\ReactCheckBoxEvent.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\@react-native-community\geolocation\android\src\main\java\com\reactnativecommunity\geolocation\GeolocationModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\react-native-create-thumbnail\android\src\main\java\com\createthumbnail\CreateThumbnailModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\react-native-create-thumbnail\android\src\main\java\com\createthumbnail\CreateThumbnailModule.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\react-native-document-picker\android\src\main\java\com\reactnativedocumentpicker\DocumentPickerModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\react-native-fbsdk-next\android\src\main\java\com\facebook\reactnative\androidsdk\Utility.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\@react-native-firebase\auth\android\src\main\java\io\invertase\firebase\auth\ReactNativeFirebaseAuthModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\@react-native-firebase\functions\android\src\main\java\io\invertase\firebase\functions\UniversalFirebaseFunctionsModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\@react-native-firebase\messaging\android\src\main\java\io\invertase\firebase\messaging\ReactNativeFirebaseMessagingModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\@react-native-google-signin\google-signin\android\src\main\java\com\reactnativegooglesignin\RNGoogleSigninModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\@react-native-google-signin\google-signin\android\src\main\java\com\reactnativegooglesignin\RNGoogleSigninModule.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\react-native-image-picker\android\src\main\java\com\imagepicker\Utils.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\react-native-localize\android\src\main\java\com\zoontek\rnlocalize\RNLocalizeModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\react-native-nodemediaclient\android\src\main\java\cn\nodemedia\react_native_nodemediaclient\NodeCameraViewManager.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
[... 106 lines cut ...]
    at makeError (C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\execa\index.js:174:9)
    at C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\execa\index.js:278:16
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async runOnAllDevices (C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\@react-native-community\cli-platform-android\build\commands\runAndroid\runOnAllDevices.js:109:5)
    at async Command.handleAction (C:\Users\zakri\Downloads\Dare-Me-App-main\node_modules\@react-native-community\cli\build\index.js:192:9)
info Run CLI with --verbose flag for more details.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
PS C:\Users\zakri\Downloads\Dare-Me-App-main>
```

### Device

Android

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

danielmark0116, 2022-12-09:
It seems like the answer for your issue might be [here - 957](https://github.com/mrousavy/react-native-vision-camera/issues/957)

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy