margelo/react-native-vision-camera #1381: How to take photo from Code (e.g.: QR Code) when detect it with frame processor (at the same time)❓

TaheriSDE, 2022-12-11:
### Question

@mrousavy 
 
Dear mrousavy 

I used to be able to take a picture of a code (e.g.: QR Code)  when it was detected with RN Camera
  But considering that RN Camera is deprecated. Please advise that such a thing can be done using react-native-vision-camera and how?

Thank you very much for your kindness

### VisionCamera Version

^2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-12-12:
Use the [VisionCamera QR Code scanner](https://github.com/rodgomesc/vision-camera-code-scanner/) fp plugin and then call `takePhoto()` using `runOnJS`

closed 2022-12-12 by mrousavy

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀 

Try upgrading to V3 if you can, it has a much more stable yet flexible foundation.

If you appreciate me dedicating my free time to improving VisionCamera and implementing features like the Code Scanner, please consider [**sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy) to show your support.