margelo/react-native-vision-camera #1382: Api upload❓

subhadeepquantiantech, 2022-12-13:
### Question

How to upload video using form data?

### VisionCamera Version

new

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

shiroze, 2022-12-28:
You can ask at Stackoverflow or search there. But there is this method using `axios` or `react-native-fs`

Axios Example :
```
const data = new FormData();
data.append('photo', {
    fileName: fileName,
    type: 'image/jpeg',
    uri: photo.path
});
axios.post(`${YOUR_API_URL}/upload-image`, data))
    .then((respJson) => {
         //Do Something here after upload finished
    })
    .catch((err) => {
        console.log(err);
    })
```

The problem is when taking snapshot/photo and get image from cache folder return file or directory not exist/found. is there any workaround sir @mrousavy

closed 2022-12-28 by mrousavy

erickhdzreyes, 2023-07-26:
> You can ask at Stackoverflow or search there. But there is this method using `axios` or `react-native-fs`
>
> Axios Example :
>
> ```
> const data = new FormData();
> data.append('photo', {
>     fileName: [...]

And hoy can I get the mime type from de file? In order to get it dynamically

LuizCristino, 2023-07-28:
The photo.path has the mime at the end. You get get it with a simple regex ou split the string by . (dot).

mrousavy, 2023-07-31:
Photo type should pretty much always be JPEG afaik