margelo/react-native-vision-camera #1389: Advise on making it work with react-native-hole-view❓

wan54, 2022-12-18:
### Question

Hi,

The main question is that is there a way in react-native-vision-camera to adjust the capturing frame size and position so that it's align with that of react-native-hole-view? Anything that is outside of the box won't be scanned.

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

andreiwow2, 2022-12-20:
I tried once to do something like this, I think the only way to achieve this would be with math to edit the image after it was taken.

I think you could make use of the frame processor to create your own algorithm.
https://mrousavy.com/react-native-vision-camera/docs/guides/frame-processors-plugins-overview

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

mrousavy, 2023-09-30:
And no, SurfaceView is not under normal view hierarchy so this will only work on iOS

closed 2023-09-30 by mrousavy