margelo/react-native-vision-camera #1392: Is there a way to add a custom View while the camera loads❓

andreiwow2, 2022-12-20:
### Question

There is a short delay until the camera is displayed and nothing is there as a placeholder, I was looking for ways to add a custom View until the camera is loaded but even onInitialized is called before the camera is visible.

So is there a way to display a placeholder until the camera is visible?

### What I tried

I tried to use onInitialized but it is called before camera is phisically visible.

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mehranjavid, 2022-12-24:
have a look at the readme of this repo [vision-camera-code-scanner](https://github.com/rodgomesc/vision-camera-code-scanner) you can place a check on device. 
`device != null ? showCamera : showPlaceholder`
if its null show your placeholder otherwise display camera

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy