margelo/react-native-vision-camera #1393: ❓ Stream video rather than record?

Sid-Turner-Ellis, 2022-12-21:
### Question

Is there any way to get access to the video stream rather than the end recording? 

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

anderpaz, 2023-04-11:
I have the same problem, I need to record the video on the device and transmit the recording in real time.
Does anyone have any suggestions or could help out?

Sid-Turner-Ellis, 2023-04-12:
Do you need audio? If not then you could look into frame processors

fukemy, 2023-06-20:
@Sid-Turner-Ellis Can u provide some way to do it? I saw Agora WebRTC can using it by this method:
```
Agora.pushWebRTCFrame(newFrame)
```

 But I dont know how it were decraled

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

JFYI; You can implement this with a native FP plugin 👍

closed 2023-09-30 by mrousavy