margelo/react-native-vision-camera #1395: [Crash] Android - AssertionException: Expected to run on UI thread!🐛

arjun-anar, 2022-12-22:
### What were you trying to do?

Trying to record a video

### Reproduceable Code

```tsx
@ReactMethod
  fun takePhoto(viewTag: Int, options: ReadableMap, promise: Promise) {
    -->  coroutineScope.launch {
      withPromise(promise) {
        val view = findCameraView(viewTag)
        view.takePhoto(options)
      }
    }
  }
```

### What happened instead?

The app crashes. Coroutine scope should be on Main Dispatcher rather than using Default Dispatcher

### Relevant log output

```shell
com.facebook.react.bridge.AssertionException: Expected to run on UI thread!
                                                                                                    	at com.facebook.react.bridge.SoftAssertions.assertCondition(SoftAssertions.java:37)
                                                                                                    	at com.facebook.react.bridge.UiThreadUtil.assertOnUiThread(UiThreadUtil.java:26)
                                                                                                    	at com.facebook.react.uimanager.UIManagerModule.resolveView(UIManagerModule.java:960)
                                                                                                    	at com.mrousavy.camera.CameraViewModule.findCameraView(CameraViewModule.kt:87)
                                                                                                    	at com.mrousavy.camera.CameraViewModule.access$findCameraView(CameraViewModule.kt:34)
                                                                                                    	at com.mrousavy.camera.CameraViewModule$startRecording$1.invokeSuspend(CameraViewModule.kt:117)
                                                                                                    	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
                                                                                                    	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
                                                                                                    	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
                                                                                                    	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:749)
                                                                                                    	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
                                                                                                    	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)
```
```

### Device

Android Emulator - SDK 33

### VisionCamera Version

^2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jinman, 2022-12-28:
We are also facing the same issue. Any pointers?

AngeloMateus, 2023-01-04:
I have a bash script run on each yarn while the maintainer is fixing this:

`//THIS temp FIXES KOTLIN SIDE NATIVE MODULE MAIN THREAD takePhoto()
const fs = require("fs");
let fileInit= fs
    .readFileSync("./node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt")
    .toString()
    .split("\n");
let file=fileInit

let index = file.findIndex((i) =>
  i.includes("internal var coroutineScope = CoroutineScope(Dispatchers.Default)")
);
if (index){
  file[index] ="internal var coroutineScope = CoroutineScope(Dispatchers.Main)"
}
if (fileInit != file.join("\n"))
  fs.writeFileSync("./node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt", file.join("\n"));
`

Then just add it to postinstall script in package.json and yarn, or just manually change that line to use Dispatchers.Main

Purushottam-industrility, 2023-01-05:
@AngeloMateus,
Above script will replace "Dispatchers.Default" with "Dispatchers.Main". Bun in my case `CameraView.kt` already contains "Dispatchers.Main". so the script will have no effect.

In my case I'm using following packages:
react-native 0.68.2
react-native-vision-camera 2.15.2
vision-camera-code-scanner 0.2.0

<Camera
              device={devices.back}
              isActive={true}
              frameProcessor={frameProcessor} // <== this line
              frameProcessorFps={5}
/>

When frameProcessor prop is passed to Camera, it throws error `AssertionException: Expected to run on UI thread` and app crashes. Do you have any other workaround on this?

jinman, 2023-01-09:
@AngeloMateus we tried above file. but not luck. Can you please let us know what is the exact difference/bug that we should fix in the post install script?

@arjun-anar any other ideas/pointers?

arjun-anar, 2023-06-26:
i was able to fix this issue by doing the following,
clone repo
goto node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt
replace `Dispatchers.Default` with `Dispatchers.Main`

or you can try upgrading to the latest version of the library and hope that it fixes it

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy