margelo/react-native-vision-camera #1396: 🐛 camera crashes when implementing tap to focus

rynell-lee, 2022-12-22:
### What were you trying to do?

I am trying to implement tap to focus feature using the focus feature and react-native-gesture-handler tap gesture function.

### Reproduceable Code

```tsx
const singleTap = Gesture.Tap()
    .maxDuration(250)
    .onStart(async (event: TapGestureHandlerEventPayload) => {
      // console.log(event);
      let point: Point = {
        x: event.x,
        y: event.y,
      };
      console.log(point);
      try {
        cameraRef.current?.focus(point);
      } catch (err) {
        console.error(err);
      }
    });

<GestureDetector gesture={Gesture.Exclusive(singleTap)}>
        <Camera
          ref={cameraRef}
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={isAppForeground}
          preset="high"
          video={true}
          audio={true}
          enableZoomGesture={true}
          onInitialized={() => SetReady(true)}
          onError={(error) => {
            console.log(error);
          }}
        />
      </GestureDetector>
```

### What happened instead?

I am able to log out the x and y coordinates showing that the tap function works, however when I implemented the focus function my camera immediately crashes. I tried catching the error but it was not logged out. 

### Device

Samsung Galaxy tab S8 (Android 12)

### VisionCamera Version

2.15.2

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

demsr, 2022-12-22:
@rynell-lee please make sure your device [supports focus](https://mrousavy.com/react-native-vision-camera/docs/api/interfaces/CameraDevice#supportsfocus)

rynell-lee, 2022-12-22:
> @rynell-lee please make sure your device [supports focus](https://mrousavy.com/react-native-vision-camera/docs/api/interfaces/CameraDevice#supportsfocus)

[image]

@demsr my device supports focus!

guiac, 2023-01-06:
same here. any solution?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

mrousavy, 2023-08-18:
Ohh actually I think in that PR I forgot to add touchToFocus, so I still need to implement that!

closed 2023-09-01 by mrousavy