margelo/react-native-vision-camera #1398: 🐛  Android build error in monorepo

Sid-Turner-Ellis, 2022-12-22:
### What were you trying to do?

Install and build React native app with react-native-vision-camera within a monorepo

### What happened instead?

Getting a gradle build error that disappears after I remove the react-native-vision package and reoccurs after I re-install it 

### Relevant log output

```shell
The Android Gradle plugin supports only kotlin-android-extensions Gradle plugin version 1.6.20 and higher.
The following dependencies do not satisfy the required version:
project ':react-native-safe-area-context' -> org.jetbrains.kotlin:kotlin-gradle-plugin:1.6.10
```

### Device

Android

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Sid-Turner-Ellis, 2022-12-23:
Fixed this by using patch package to set the kotlin version to 1.6.20 instead of 1.5.x, closing it because I don't think its related to monorepos rather than the packages I had in my monorepo

Sid-Turner-Ellis, 2022-12-24:
Re-opening this as though its stopping the build error my app is crashing whenever the camera opens

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy