margelo/react-native-vision-camera #1399: ❓ react-native-vision-camera doesn't appear to work in Storybook

marcodafonseca, 2022-12-23:
### Question

I'm trying to test a component that utilizes your camera component.

However, in Storybook, the useCameraDevices(); hook always returns:
```
{
    "back": undefined, 
    "external": undefined, 
    "front": undefined, 
    "unspecified": undefined
}
```

### What I tried

Setup Storybook and tried setting up a story for my component

### VisionCamera Version

^2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

marcodafonseca, 2022-12-23:
This is not an issue. I made a mistake in how I was using the tool.

closed 2022-12-23 by marcodafonseca

fullsnack-DEV, 2022-12-28:
I also get the same Error. How did you solve it?

marcodafonseca, 2022-12-28:
Honestly? I didn't. I removed Storybook (as importing it was causing my issues for some reason) from my solution and did my testing in a far less convenient way.