margelo/react-native-vision-camera #1400: 🐛 getCameraPermissionStatus never resolves

marcodafonseca, 2022-12-23:
### What were you trying to do?

I am trying to use the Camera component

### Reproduceable Code

```tsx
import { storiesOf } from "@storybook/react-native";
import React, { useEffect, useState } from "react";
import { Button } from "../../controls/button";
import { Camera as VisionCamera } from "react-native-vision-camera";

function SimpleRepro() {
    const [showCamera, setShowCamera] = useState(true);

    const checkCameraPermissions = async () => {
        try {
            console.log("checkCameraPermissions - start");

            var permissionStatus =
                await VisionCamera.getCameraPermissionStatus();
            console.log("checkCameraPermissions", permissionStatus);

            switch (permissionStatus) {
                case "authorized":
                    setShowCamera(true);
                    break;
                default:
                    var requestPermissionStatus =
                        await VisionCamera.requestCameraPermission();
                    switch (requestPermissionStatus) {
                        case "authorized":
                            setShowCamera(true);
                            break;
                        case "denied":
                        default:
                            console.log("denied", requestPermissionStatus);
                    }
            }
        } catch (error) {
            console.log(error);
        }
    };

    useEffect(() => {
        checkCameraPermissions();
    }, []);

    if (showCamera) return <></>;

    return (
        <Button
            label="Show Camera"
            onPress={() => setShowCamera(true)}></Button>
    );
}

```
```

### What happened instead?

My screen remains blank.
I can't see any errors being thrown anywhere (in metro, or console, or in Logcat)

The only thing seeminly being logged in console is "checkCameraPermissions - start", so it looks like getCameraPermissionStatus never resolves.

### Relevant log output

```shell
LOG  Running "Redacted" with {"rootTag":1}
 LOG  checkCameraPermissions - start
```

### Device

Pixel 4 Emulator (Android v10)

### VisionCamera Version

^2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mehranjavid, 2022-12-23:
I have tried running your code and these are the Logs that were printer on my console: 
i am using  "react-native-vision-camera": "^2.15.2",

when camera was already granted permission
 LOG  checkCameraPermissions - start
 LOG  checkCameraPermissions authorized
 
 when app was freshly installed
 LOG  checkCameraPermissions - start
 LOG  checkCameraPermissions denied

marcodafonseca, 2022-12-23:
I suspect my problem is irrespective of this plugin maybe.

I changed my "checkCameraPermissions" function to the following and I still can't get my promise to resolve nor am I seeing any errors anywhere:

```
const checkCameraPermissions = async () => {
    console.info("PermissionsAndroid");

    var _resolve: any;

    var testPromise = new Promise<boolean>((resolve) => {
        _resolve = resolve;
    });

    setTimeout(() => {
        _resolve(true);
        console.info("timeout");
    }, 1000);

    var resultFake = await testPromise;

    console.info(resultFake);
};

```

closed 2022-12-24 by marcodafonseca

smallzZz8, 2023-01-17:
Same.. What was your solution?

marcodafonseca, 2023-01-19:
> Same.. What was your solution?

I wouldn't say I "resolved" the issue, per se. I got frustrated and started removing libraries to see what was causing the problem.
Once I removed Storybook everything started working correctly again.

I have Storybook working without issue in other Native projects, so I wouldn't blame the library. I just didn't have time to figure out what about my implementation was causing problems and outright removed Storybook from my project.