margelo/react-native-vision-camera #1401: 🐛 take photo returns photo file with width and height flipped

Robogram, 2022-12-27:
### What were you trying to do?

I have been struggling with cropping the image file after 

let { path, width, height } = await camera.current.takePhoto({
      qualityPrioritization: "speed",
      skipMetadata: true,
      flash: "off"
    })

### What happened instead?

The width and height is flipped. width is height and height is width.

### Device

iPhone 11

### VisionCamera Version

^2.15.2

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

lucipacurar, 2023-01-06:
I have the same problem and it triggers a lot of other issues: we're doing some auto cropping and that is broken because of this, and in some parts of our backend the images are flipped when adding them to a PDF.

Robogram, 2023-01-15:
and also, i have two android phones, the front taking photo is flipped in one and not the other

lucipacurar, 2023-01-15:
As a workaround I'm using https://github.com/bamlab/react-native-image-resizer to resize the image to the same size and it fixes the orientation issues.

Robogram, 2023-01-16:
const photo = await camera.current.takePhoto({
      qualityPrioritization: "speed",
      skipMetadata: true,
      flash: "off",
    })
let path = photo.path, width = photo.width, height = photo.height
    let uri = Platform.OS == "ios" ? path : `file://${path}`

ImageResizer.createResizedImage(
		uri,
		width,
		height,
		"JPEG",
		100,
		0,
		undefined,
		false
)
	.then(async(response) => {
})
	  .catch((err) => {
	    // Oops, something went wrong. Check that the filename is correct and
	    // inspect err to get more details.
	    console.log(err, path)
	  });

ios logs: [Error: Image uri invalid.] /private/var/mobile/Containers/Data/Application/324E4E06-EF5C-4F5B-A4EF-5FD7B89E607D/tmp/ReactNative/EBDB1F6D-43A5-404B-875D-DF68C7FFB64D.jpeg

android: doesn't fix orientation

any help would be great thanks:)

lucipacurar, 2023-01-16:
We used destination directory as well. 

```
const compressedImage = await ImageResizer.createResizedImage(
					`file://${photo.path}`,
					photo.height, // width
					photo.width, // height
					'JPEG',
					80, // compression
					0, // rotation
					destinationDirectory,
				);
```

Robogram, 2023-01-16:
When I set destination directory to undefined, it works.

This is what I did before, 
let destinationDirectory = FileSystem.documentDirectory + "/dsjfkldsfjsdlkjf.jpeg"
outputs: [Error: Image uri invalid.]

maybe the image uri is private....

Robogram, 2023-01-16:
So you're using the same directory as the original?

lucipacurar, 2023-01-16:
Yes, the same directory. We just extract it from `photo.path`:
```TypeScript
const photoPathArray = photo.path.split('/');
				photoPathArray.pop();
```

Robogram, 2023-01-16:
Found something better:)

let flipImage = false

if (photo.height > photo.width) {
flipImage = true
}

Robogram, 2023-01-23:
Hey there, something weird I found,

(devices.front) && (
	            <Camera
	              style={{ height: '100%', width: '100%' }}
	              ref={camera}
	              device={devices.front}
	              isActive photo video audio
	            />
	          )

When I take a photo at first, it captures. But when I record a video and then try taking a photo again, it doesn't capture.......

So I had to do this for now
(devices.front) && (
	            <Camera
	              style={{ height: '100%', width: '100%' }}
	              ref={camera}
	              device={devices.front}
	              isActive photo video={recording} audio={recording}
	            />
	          )

'recording' is set to false as default, when the user press recording and before it starts, I would set recording to true and put a 0.5 seconds timeout before starting the recording process, after I press stop recording, then I set the recording state back to false.

I hope you understand what I'm saying.

Cheers:)

EwaGuziejko, 2023-02-22:
> As a workaround I'm using https://github.com/bamlab/react-native-image-resizer to resize the image to the same size and it fixes the orientation issues.

I tried the same but it messes up with photo dimensions :/ it didn't happen for you lucipacurar?

lucipacurar, 2023-02-23:
No we didn't have problems yet with the dimensions. We tested on our test lab devices and we didn't receive any weird customer reports yet. But please have a second look at the code above where we switch the width for the height and vice versa.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

JFYI; Camera sensor orientation is 90deg, maybe thats why it's flipped.

closed 2023-09-30 by mrousavy