margelo/react-native-vision-camera #1403: White screen before camera on 🐛

ChauVV, 2022-12-29:
### What were you trying to do?

I get a white screen in 3s before camera on.
```
            {device != null && hasPermission && (
                <Camera
                  ref={cameraRef}
                  onInitialized={showPopupInstruction}
                  style={StyleSheet.absoluteFill}
                  device={device}
                  isActive
                  orientation="portrait"
                  focusable
                  enableDepthData
                  photo
                />
             )}
```

### What happened instead?

I get a white screen in 3s before camera on.

### Device

android real device

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

shakuniyamihir8400, 2022-12-30:
Having the same issue here, Any update please

parth-koshta, 2023-04-11:
@ChauVV @shakuniyamihir8400 
Hey guys, I'm facing the same issue on some devices. Were you able to find a workaround for this?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

mrousavy, 2023-08-18:
The white screen might be because your View is white. Add backgroundColor: black to your view

closed 2023-09-01 by mrousavy