margelo/react-native-vision-camera #1405: Camera loading speed is slow🐛

shakuniyamihir8400, 2022-12-30:
### What were you trying to do?

When I reach to that particular screen where camera is placed, it takes 2-3 sec to load camera.
How I can load fast camera comp.

### What happened instead?

It takes 2-3 sec to load camera.

### Device

Android

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

kkj0712, 2023-02-22:
You still have this problem? I'm facing this issue..
What kind of device did you test?

lucksp, 2023-03-14:
i am also seeing this..  

https://user-images.githubusercontent.com/15162169/225092336-86f0017e-8a9b-4e6f-a721-335e1b93e0da.MP4

lucksp, 2023-03-14:
I did find this suggestion, which I implemented...it's a little better:

https://github.com/mrousavy/react-native-vision-camera/discussions/723#discussioncomment-1938667

I still am wondering why no children are rendered...

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

Check the docs for more information on how Camera Devices work to find an explanation

closed 2023-09-30 by mrousavy

duongthai910, 2023-11-16:
Did you fix this? I am now at the latest version 3.6.6 but still facing this issue...

FLoda96, 2023-11-25:
Same thing for me on Android with a OnePlus North, i tried to use the suggestion of the wide-angle but apparently only front and back are recognized as valid type of cameras