margelo/react-native-vision-camera #1406: 🐛[android] navigation problem

baesumin, 2023-01-01:
### What were you trying to do?

I come back from another screen and try to use the camera again.

### What happened instead?

I see only a black screen. But the camera function works.
(If i take a picture on a black screen, the actual screen is taken.)

### Device

s21+ ultra

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ramssutharr, 2023-01-04:
having same issue. are you able to resolved it ?

baesumin, 2023-01-10:
> Hi !
> 
> I think the problem will be fix in V3 (#1251).
> 
> As far I'm concerned, I "fixed" the issue by adding a condition on isActive, depending on isFocused props on react-navigation. It forces the Camera to be reload. It's not ideal, but it's better than nothing until V3 is ready :-)
thanks! :)

closed 2023-01-10 by baesumin

muzhaqi16, 2023-01-24:
I am still having this issue on Android with react-nativgation. I am using the isFocused hook but it still persists.

baesumin, 2023-01-24:
> I am still having this issue on Android with react-nativgation. I am using the isFocused hook but it still persists.

i resolved it.

const isfocused = useIsFocused();
<Camera isActive={isfocused} .../>

dunapanta, 2023-03-10:
For me it works with @baesumin solution but also if I use useNavigation hook instead of navigation

`const {navigate} = useNavigation();`
```
navigate('ScreenName', {
          prop
        })
```

jasbirrajrana, 2023-03-29:
still having this issue  I am using the isFocused hook but it still persists.