margelo/react-native-vision-camera #1407: 🐛 Unable to use external USB camera

rawatnaresh, 2023-01-04:
### What were you trying to do?

I wanted to use external USB camera in my android app to take photo. 

```js
const devices = useCameraDevices();
console.log(devices.external); // returns undefined although my usb camera is connected
```

### Reproduceable Code

```tsx
import React, {useCallback, useEffect, useState} from 'react';
import {Button, View, ToastAndroid} from 'react-native';
import {
  Camera,
  useCameraDevices,
  CameraRuntimeError,
} from 'react-native-vision-camera';

const App = () => {
  const devices = useCameraDevices();
  const [cameraPosition, setCameraPosition] = useState<
    'front' | 'back' | 'external'
  >('back');
  const [isCameraInitialized, setIsCameraInitialized] = useState(false);

  const device = devices[cameraPosition];

  useEffect(() => {
    const requestPermissions = async () => {
      await Camera.requestCameraPermission();
    };
    requestPermissions();
  });

  // Camera callbacks
  const onError = useCallback((error: CameraRuntimeError) => {
    console.error(error);
  }, []);

  const onInitialized = useCallback(async () => {
    setIsCameraInitialized(true);
  }, []);

  return (
    <>
      <View style={{flexDirection: 'row'}}>
        <Button
          title="Use USB Camera"
          onPress={() => {
            setCameraPosition('external');
            ToastAndroid.showWithGravity(
              'External camera enabled',
              ToastAndroid.SHORT,
              ToastAndroid.BOTTOM,
            );
          }}
        />
        <Button
          title="Use back Camera"
          onPress={() => {
            setCameraPosition('back');
            ToastAndroid.showWithGravity(
              'Back camera enabled',
              ToastAndroid.SHORT,
              ToastAndroid.BOTTOM,
            );
          }}
        />
      </View>
      {device != null && (
        <Camera
          device={device}
          isActive={isCameraInitialized}
          onInitialized={onInitialized}
          onError={onError}
          photo={true}
          video={false}
          style={{height: '90%'}}
        />
      )}
    </>
  );
};

export default App;
```

### What happened instead?

External USB camera was not detected. Only `front` and `back` are available. It doesn't even ask for usb permissions like other apps.

### Vision camera demo
https://user-images.githubusercontent.com/29520305/210509978-a856a705-a52c-441d-9c7d-82b39332aa8c.mp4

I can see my usb camera when I use it with other app

### [Android USB camera](https://github.com/jiangdongguo/AndroidUSBCamera)
https://user-images.githubusercontent.com/29520305/210510747-155ad242-5a80-48f9-8255-fc60a8423b33.mp4

### Device

All android devices

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

a1029382825, 2023-04-04:
I am also facing same problem with this package. Please suggest us something

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

mrousavy, 2023-08-18:
Also, can you add

```tsx
Camera.getAvailableCameraDevices().then((d) => console.log(JSON.stringify(d)))
```

to your code and then paste the output of that here? I'd like to see what devices it detects.

bhimeshchauhan, 2023-08-22:
@mrousavy I tried the line above but it returns me `undefined` with the USB device ( albeit it could be an issue with the manufacturer itself, seemed like I might have to use USB API from Android to create a custom module to support such functionality, I'd be more than happy to try and add it to the library if it doesn't already exist )

mrousavy, 2023-08-24:
@bhimeshchauhan I unfortunately don't have an USB camera to test this, but can you maybe try the official Google camera2 example app to see if it even works with Android's Camera API?

1. Download https://github.com/android/camera-samples
2. Open `Camera2Basic` **or** `Camera2Video` in Android Studio
3. Run it on your device and see if the device is available in there (I think in video you should be able to select any camera device, not sure)

---

Also, in VisionCamera it should log all devices to the logcat logs (`adb logcat`/Android Studio) that become available. Maybe you could double-check there if it is availble?

Can you even use the USB Camera in the stock Camera app on your phone?

mrousavy, 2023-08-24:
Ah, see [this StackOverflow post](https://stackoverflow.com/a/68120774/5281431).

closed 2023-09-01 by mrousavy

tiste, 2023-09-19:
Hello all. With current iOS17, we should be able to use external camera on iPhone/iPad. Do you know if « external » camera mode could be working with this library?
Thanks a lot.

mrousavy, 2023-09-19:
Hey @tiste, you made me curious. Look at this PR: https://github.com/mrousavy/react-native-vision-camera/pull/1813 😄