margelo/react-native-vision-camera #1412: ❓ Digital Zoom

hocaltar, 2023-01-08:
### Question

First, great job with this project.

Problem: I can't get my zoom to go more than the max optical zoom of the device... I am using a Samsung Galaxy S22 Ultra and device is 100% utilizing the telephoto lens. I want to be able to digitally zoom to 20 power, or double that of the device's optical zoom.
 

### What I tried

For instance: I zoom in with my pinch gesture (minZoom = **10**;  maxZoom = **20**;) 20 is 10 more than the device's optical maxZoom in the devices' format. It does nothing... The optical maxZoom seems to be hard-coded...

Any ideas or help is much appreciated!

### VisionCamera Version

V3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

hocaltar, 2023-01-10:
Still struggling with this, I’m thinking it’s a Samsung issue… Can anyone confirm that they can zoom more than the optical maxZoom of their device?

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

mrousavy, 2023-09-30:
It's probably a Samsung issue tbh 😓

closed 2023-09-30 by mrousavy