margelo/react-native-vision-camera #1413: 🐛 iOS 'react/bridging/CallbackWrapper.h' file not found

salihgun, 2023-01-09:
### What were you trying to do?

I was trying to add the library but can not run it

### What happened instead?

[image]

### Device

iPhone 14 (iOS 16.0)

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

luchozamora, 2023-02-06:
@salihgun make sure you are using `SWIFT_VERSION = 5.2` in your project.
From react-native-camera's [troubleshooting website](https://mrousavy.com/react-native-vision-camera/docs/guides/troubleshooting):
```
3. iii. If the SWIFT_VERSION value is set, make sure it is set to 5.2 or higher.
```

senrymori, 2023-03-26:
> @salihgun make sure you are using `SWIFT_VERSION = 5.2` in your project. From react-native-camera's [troubleshooting website](https://mrousavy.com/react-native-vision-camera/docs/guides/troubleshooting):
> 
> ```
> 3. iii. If the SWIFT_VERSION value is set, make sure it is set to 5.2 or higher.
> ```

i raised the swift version to 5.2 but it doesn't work (xCode v14.2)

but this answer helped me: https://github.com/mrousavy/react-native-vision-camera/issues/1159#issuecomment-1192621018

when I raised the version to 5.2 and registered this in my Podfile, then the whole project was assembled

```ruby
post_install do |installer|
    installer.pods_project.targets.each do |target|
      target.build_configurations.each do |config|
        config.build_settings['EXPANDED_CODE_SIGN_IDENTITY'] = ""
        config.build_settings['CODE_SIGNING_REQUIRED'] = "NO"
        config.build_settings['CODE_SIGNING_ALLOWED'] = "NO"
        config.build_settings['HEADER_SEARCH_PATHS'] ||= '$(inherited) '
        config.build_settings['HEADER_SEARCH_PATHS'] << '"$(PODS_ROOT)/Headers/Private/React-bridging/react/bridging" '
        config.build_settings['HEADER_SEARCH_PATHS'] << '"$(PODS_CONFIGURATION_BUILD_DIR)/React-bridging/react_bridging.framework/Headers" '    
       end
    end
  end
```

kesha-antonov, 2023-06-23:
> ```ruby
> .build_settings['HEADER_SEARCH_PATHS'] << '"$(PODS_ROOT)/Headers/Private/React-bridging/react/bridging" '
>         config.build_settings['HEADER_SEARCH_PATHS'] << '"$(PODS_CONFIGURATION_BUILD_DIR)/React-bridging/react_bridging.framework/Headers" '
> ```

In RN 0.72 paths changed. Here's fix:

```
post_install do |installer|
  installer.pods_project.targets.each do |target|
    target.build_configurations.each do |config|
      config.build_settings['HEADER_SEARCH_PATHS'] ||= '$(inherited) '
      config.build_settings['HEADER_SEARCH_PATHS'] << '"${PODS_ROOT}/../../node_modules/react-native/ReactCommon" '
    end
  end
end
```

RRaideRR, 2023-06-29:
Small reproducible example - don't know if it helps you @mrousavy 

https://github.com/RRaideRR/vision-camera-expo-sdk-49

I'm seeing this only when I enable "useFrameworks": "static"

mrousavy, 2023-06-29:
Hey thanks!
I'm not sure if VisionCamera can build with static frameworks, let me investigate that in v3. Thanks for the repro!

RRaideRR, 2023-06-29:
> Hey thanks! I'm not sure if VisionCamera can build with static frameworks, let me investigate that in v3. Thanks for the repro!

before react-native 0.72.0 it did :-)

kbrandwijk, 2023-07-10:
Are there any plans to fix this in the current VisionCamera version?

mrousavy, 2023-07-10:
Hey @kbrandwijk - unfortunately not, no. I just can't focus on everything - I'm trying to plan my time as effective as possible and in V3 I'll do this - hope you understand

kbrandwijk, 2023-07-10:
A little note in the readme that RN 0.71 is the last supported version might help in that case...

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

jackylu0124, 2024-04-15:
> In RN 0.72 paths changed. Here's fix:
>
> ```
> post_install do |installer|
>   installer.pods_project.targets.each do |target|
>     target.build_configurations.each do |config| [...]

@kesha-antonov I have a quick question: should I put the code you mentioned before or after the `react_native_post_install()` and `__apply_Xcode_12_5_M1_post_install_workaround()` calls?

In other words, in my `Podfile`, should I do
```
post_install do |installer|
  installer.pods_project.targets.each do |target|
    target.build_configurations.each do |config|
      config.build_settings['HEADER_SEARCH_PATHS'] ||= '$(inherited) '
      config.build_settings['HEADER_SEARCH_PATHS'] << '"${PODS_ROOT}/../../node_modules/react-native/ReactCommon" '
    end
  end

  react_native_post_install(
    installer,
    config[:reactNativePath],
    :mac_catalyst_enabled => false
  )
  __apply_Xcode_12_5_M1_post_install_workaround(installer)
end
```

or should I do

```
post_install do |installer|
  react_native_post_install(
    installer,
    config[:reactNativePath],
    :mac_catalyst_enabled => false
  )
  __apply_Xcode_12_5_M1_post_install_workaround(installer)

  installer.pods_project.targets.each do |target|
    target.build_configurations.each do |config|
      config.build_settings['HEADER_SEARCH_PATHS'] ||= '$(inherited) '
      config.build_settings['HEADER_SEARCH_PATHS'] << '"${PODS_ROOT}/../../node_modules/react-native/ReactCommon" '
    end
  end
end
```

Thanks a lot in advance!