margelo/react-native-vision-camera #1414: Is there a way to select the audio recording device❓

fnperez-ilanthealth, 2023-01-09:
### Question

I'm building a recording app which is intended to be able to record video/audio with high quality devices. 
What I need is the possibility of selecting the audio recording device (such as Focurite 2i2), but I'm not being able. 

Is there a way I can accomplish this? Or does anyone know how to help me find a solution?

Thanks

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

Kyrluckechuck, 2023-10-25:
@mrousavy can this be reopened since this functionality was not added in v3 and is not something currently possible?

mrousavy, 2023-10-31:
Hey - I know for a fact that on iOS it is not possible to do this at the moment. You can only select a mic.

So this needs third party drivers, which is not the responsibility of VisionCamera