margelo/react-native-vision-camera #1415: 🐛 Using qualityPrioritization = 'quality' will crash the app

andreiwow2, 2023-01-09:
### What were you trying to do?

I tried to take a picture while using qualityPriuoritization set to quality.

### Reproduceable Code

```tsx
const takePhoto = async () => {
    try {
      if (cameraRef.current) {
        const centerPoint = {
          x: Dimensions.get('window').width / 2,
          y: Dimensions.get('window').height / 2,
        };
        await cameraRef.current.focus(centerPoint);
        const photo = await cameraRef.current.takePhoto({
          flash: useFlash ? 'on' : 'off',
          qualityPrioritization: 'quality',
        });
        toast.close(takingPhotoToastRef.current);
        setTakenPhoto(photo);
        takingPhotoToastRef.current = toast.show({
          title: 'Foto aufgenommen',
          placement: 'top',
          duration: 3000,
        });
        setCurrentlyTakingPhoto(false);
      }
    } catch (err) {
      console.log(err);
      takingPhotoToastRef.current = toast.show({
        title: 'Foto aufgenommen',
        placement: 'top',
        description: `${err}`,
        duration: 3000,
      });
    }
  };
```

### What happened instead?

App crashed with error in xCode

### Relevant log output

```shell
flipper: FlipperClient::addPlugin Inspector
flipper: FlipperClient::addPlugin Preferences
flipper: FlipperClient::addPlugin React
flipper: FlipperClient::addPlugin Network
2023-01-09 17:02:48.218518+0100 KareOcrBusinessCards[24531:3984908] [] nw_endpoint_copy_association_with_evaluator nw_context_caches_are_shared(context (<nw_context com.apple.CFNetwork.NSURLSession.{C76AB4EF-317C-4DDF-A320-B29DF9620826}{(null)}{Y}{2} (private)>), endpoint->context (<nw_context Default Network Context (private)>)) is false
2023-01-09 17:02:48.219570+0100 KareOcrBusinessCards[24531:3984908] [connection] nw_socket_handle_socket_event [C1.1.1:1] Socket SO_ERROR [61: Connection refused]
2023-01-09 17:02:48.220978+0100 KareOcrBusinessCards[24531:3984908] [connection] nw_socket_handle_socket_event [C1.1.2:1] Socket SO_ERROR [61: Connection refused]
2023-01-09 17:02:48.241201+0100 KareOcrBusinessCards[24531:3984907] [connection] nw_connection_get_connected_socket_block_invoke [C1] Client called nw_connection_get_connected_socket on unconnected nw_connection
2023-01-09 17:02:48.241233+0100 KareOcrBusinessCards[24531:3984907] TCP Conn 0x28109c6e0 Failed : error 0:61 [61]
2023-01-09 17:02:48.270815+0100 KareOcrBusinessCards[24531:3984908] Invalid encryption level
2023-01-09 17:02:48.288751+0100 KareOcrBusinessCards[24531:3984908] Invalid encryption level
2023-01-09 17:02:48.288823+0100 KareOcrBusinessCards[24531:3984908] Invalid encryption level
2023-01-09 17:02:48.303830+0100 KareOcrBusinessCards[24531:3984908] [] nw_protocol_default_input_finished called with null protocol->default_input_handler
2023-01-09 17:02:48.307947+0100 KareOcrBusinessCards[24531:3984908] [] nw_protocol_default_input_finished called with null protocol->default_input_handler, dumping backtrace:
        [arm64] libnetcore-3013.42.2
    0   Network                             0x00000001b906ed44 __nw_create_backtrace_string + 192
    1   Network                             0x00000001b8cbc630 3BF445F9-D58F-3280-B9C9-2FEAF4DACA6D + 2762288
    2   Network                             0x00000001b927ca6c 3BF445F9-D58F-3280-B9C9-2FEAF4DACA6D + 8792684
    3   Network                             0x00000001b8e57e5c 3BF445F9-D58F-3280-B9C9-2FEAF4DACA6D + 4447836
    4   Network                             0x00000001b923f1b0 3BF445F9-D58F-3280-B9C9-2FEAF4DACA6D + 8540592
    5   Network                             0x00000001b923e908 3BF445F9-D58F-3280-B9C9-2FEAF4DACA6D + 8538376
    6   Network                             0x00000001b910a444 3BF445F9-D58F-3280-B9C9-2FEAF4DACA6D + 7275588
    7   Network                             0x00000001b924b1a8 3BF445F9-D58F-3280-B9C9-2FEAF4DACA6D + 8589736
[... 130 lines cut ...]
rsocket::StreamInterruptedException: connection error
2023-01-09 17:02:57.559472+0100 KareOcrBusinessCards[24531:3984904] [native] VisionCamera.takePhoto(options:promise:): Capturing photo...
2023-01-09 17:02:57.561235+0100 KareOcrBusinessCards[24531:3984904] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[AVCapturePhotoOutput capturePhotoWithSettings:delegate:] settings.photoQualityPrioritization must not be higher than self.maxPhotoQualityPrioritization'
*** First throw call stack:
(0x1b863de88 0x1b196b8d8 0x1d1690df4 0x1058faabc 0x1058fcc0c 0x10874c5a8 0x10874e05c 0x10875610c 0x108756e34 0x108763cbc 0x205697df8 0x205697b98)
libc++abi: terminating with uncaught exception of type NSException
terminating with uncaught exception of type NSException
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[AVCapturePhotoOutput capturePhotoWithSettings:delegate:] settings.photoQualityPrioritization must not be higher than self.maxPhotoQualityPrioritization'
(lldb)
```

### Device

iPhone 12 iOS 16.1.2

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mcaglarkeskin, 2023-06-21:
Just give `enableHighQualityPhotos={true}` to camera component. This will solve the problem.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

skizzo, 2023-10-17:
Unfortunately, XCode 15 is everthing but stable when working with RN 0.71.x, so updating to v3 is not an option for me (and I guess many other devs)..

PierreCapo, 2023-11-23:
FYI with vision-camera V3: 
The crash still occurs on iOS only if `qualityPrioritization: 'quality'` is set. As per mentioned by @mcaglarkeskin , setting `enableHighQualityPhotos={true}`fixes the issue.