margelo/react-native-vision-camera #1416: Camera is not loading on certain devices.

xtl-geiger, 2023-01-11:
### What were you trying to do?

The Camera is working fine on most devices. However, there are a few on which it just won't work. The camera will simply try to load and stay black.

S22 Ultra von Samsung | SM-S908B | Android 13
Huawei |  MAR-LX1B | Android 10

It might be related to https://github.com/mrousavy/react-native-vision-camera/issues/238

### What happened instead?

The camera did not load correctly.

### Relevant log output

```shell
Sadly, I have no logs seeing as I do not own the phones mentioned and have no access to these phones, yet.
```

### Device

As mentioned above

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

hocaltar, 2023-01-11:
Try disabling frame processors. Also, see which device and formats it’s using. I can confirm that it works on the S22 ultra.

ManarHussein, 2023-01-31:
@xtl-geiger are you rendering it inside Modal?

EwaGuziejko, 2023-02-08:
I was using  useCameraDevices('wide-angle-camera') and it is returning only front camera on S22 Ultra, other ones are undefined. So @xtl-geiger maybe it is the same case for you? useCameraDevices() solves it for me

powerpivar, 2023-02-27:
> Try disabling frame processors. Also, see which device and formats it’s using. I can confirm that it works on the S22 ultra.

What camera device and format are you using?

OtayNacef, 2023-05-27:
> I was using  useCameraDevices('wide-angle-camera') and it is returning only front camera on S22 Ultra, other ones are undefined. So @xtl-geiger maybe it is the same case for you? useCameraDevices() solves it for me

Same on S23 Ultra

tatianavazquez, 2023-06-18:
@ManarHussein hey! I have the same issue and I am rendering the camera view in a modal. Mine works fine for other android devices, but we have this problem with the  Xiaomi Redmi 9A, Android Version 10. If you click on the camera, it just shows a loader and nothing else.
I'm also using useCameraDevices() but I don't pass anything.
Also, for the modal, I implemented this hook: = gestureHandlerRootHOC() because otherwise it wasn't zooming.
Any thoughts?

OtayNacef, 2023-07-16:
To resolve the problem, you can simply remove the 'wide-angle-camera' parameter, like this:
const devices = useCameraDevices();"

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy

alexvazquez, 2023-11-08:
> Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).
> 
> I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?
> 
> Here's an instruction on how you can test that: [#1674 (comment)](https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217)
> 
> If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 Otherwise please let me know and I'll keep this issue open to keep track of it.
> 
> Thank you!

Hi, I have same issues with some phones not opening camera, how can I npm install the new V3 version? Do i have to change code or syntax for this rewrite? 

Appreciate so much.

Alejandro