margelo/react-native-vision-camera #1417: 🐛 signal 11 (SIGSEGV), code 2 (SEGV_ACCERR) when trying to read shared value

oscar-b, 2023-01-12:
### What were you trying to do?

I want to access shared values using `useDerivedValue`.

### Reproduceable Code

```tsx
type ScannedCode = {
	code: string;
	field?: string;
};

const codeCache = useSharedValue<ScannedCode[]>([]);

const tester = useFrameProcessor((frame) => {
	'worklet';

	try {
		const test = codeCache.value.slice();
		codeCache.value = test;
	} catch (error) {
		console.error(error);
	}
}, []);

//
// Removing this call stops the crash from happening
// 
const test = useDerivedValue(() =>
	codeCache.value.map((c) => c.field ?? ''),
);
```

### What happened instead?

The app crashes without any errors, but a segfault in logcat logs.

### Relevant log output

```shell
01-12 18:03:06.273 12773 13004 F libc    : Fatal signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0x7365fb2d7c in tid 13004 (pool-13-thread-), pid 12773
```

### Device

Android Motorola Defy

### VisionCamera Version

2.15.2

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

oscar-b, 2023-01-16:
It seems that using `object` or `array` for shared values in the frame processor (possibly when using multiple ones) leads to these crashes. I worked around it by storing the object as a string using `JSON.stringify()` and `JSON.parse()` in the frame processor.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

mrousavy, 2023-09-30:
Use `useSharedValue` from RN Worklets Core in V3

closed 2023-09-30 by mrousavy