margelo/react-native-vision-camera #1421: 🐛  camera view is showing under modal even though used inside modal component

kyimoemin, 2023-01-17:
### What were you trying to do?

I want to use `Camera` component inside `Modal` component but the camera view is showing under the Modal component. zIndex does not help either.

### excepted behavior
- show camera view inside modal view

### Reproduceable Code

```tsx
...
<Modal>
<Camera/>
</Modal>
...
```

### What happened instead?

the camera view showing under modal component.the 

### Device

Redmi Note 7 Pro

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ManarHussein, 2023-01-31:
@kyimoemin I had same issue, camera is not showing inside the modal

did you fixed it or find a workaround?

kyimoemin, 2023-02-02:
@ManarHussein try adding the exact width and height or use `StyleSheet.absoluteFill` style property to the camera component.

ManarHussein, 2023-02-02:
@kyimoemin  it's fixed with you this way, right? 
it's worked on ios simulators, but not android

kyimoemin, 2023-02-02:
to be precise, I set the parent view exact width and height, and the camera view to 100% width and height.

ManarHussein, 2023-02-02:
@kyimoemin 
> to be precise, I set the parent view exact width and height, and the camera view to 100% width and height.

by parent view you mean the modal?

ChenPan311, 2023-03-01:
same here... any solution?

lccoronel, 2023-03-31:
Same problem here 😒

Akash-Hubspire, 2023-04-09:
I was able to view the camera inside the modal, but my issue is with the back button inside the modal, it'll work only on the second press, doesn't matter if the first press is on the button or on the camera component

`import React, { useRef, useState } from 'react';
import {
  StyleSheet, Modal, Image, TouchableOpacity, View,
} from 'react-native';
import { SafeAreaView } from 'react-native-safe-area-context';
import { Camera, useCameraDevices } from 'react-native-vision-camera';

const BACK_ARROW = require('../../assets/Icon/backArrow.png');

const CameraGallery = ({
  handleImageCapture, enableCamera, onPressBack,
}) => {
  const cameraRef = useRef()
  const [cameraPosition, setCameraPosition] = useState('back');
  const devices = useCameraDevices();
  const device = devices[cameraPosition];

  const handlePressBack = () => {
    console.log('-clicked-');
    onPressBack(false);
  };

  console.log(enableCamera, '--enableCamera');

  return (
    <Modal visible={enableCamera} transparent>
      <SafeAreaView style={styles.container}>
        <TouchableOpacity
          onPress={handlePressBack}
          style={styles.backButton}
        >
          <Image
            source={BACK_ARROW}
            style={styles.backIcon}
          />
        </TouchableOpacity>
        {device != null && (
          <View style={{flex: 1 }} >
            <Camera
              ref={cameraRef}
              style={StyleSheet.absoluteFill}
              device={device}
              isActive={enableCamera}
            />
          </View>
        )}
      </SafeAreaView>
    </Modal>
  );
};

const styles = StyleSheet.create({
  container: { flex: 1, zIndex: 20 },
  backButton: {
    backgroundColor: 'red',
    position: 'absolute',
    width: 23,
    height: 23,
    zIndex: 999,
  },
  backIcon: { width: 23, height: 23, tintColor: '#FFFFFF' },
});

export default CameraGallery;`

bibekgurunguh, 2023-04-13:
Yeah problem here as well. `overflow: 'hidden'` also not working on it.

maksym-bezditko, 2023-05-03:
[kyimoemin](https://github.com/kyimoemin)

observing it on Android 11. What is your version?

Rehankhalil462, 2023-07-07:
Yes. same problem. Not working

llassan, 2023-07-13:
I solved it by enclosing camera inside Modal. 
`<Modal visible={recordVideo} transparent animationType="fade" onRequestClose={closeCamera}>
      <Camera
        style={StyleSheet.absoluteFill}
        device={device}
        ref={cameraRef}
        isActive={true}
        video={true}
        audio={true}
      />
</Modal>`

mrousavy, 2023-09-30:
Modals cannot render SurfaceViews

closed 2023-09-30 by mrousavy