margelo/react-native-vision-camera #1423: 🐛[android] Camera.requestCameraPermission() does not resolve after giving camera permissions.

jlemein, 2023-01-20:
### What were you trying to do?

The following are my reproduction steps for Android (phone and emulator).

1. I request for camera permission by calling `await Camera.requestCameraPermission();`. 
2. A popup dialog is shown asking me to decide on which permission to give. I answer "only this time".
3. The popup dialog disappears but I have no access to the camera.
4.   * Also if you look at the logs, it does not print "Permission given: ...".
5. If I restart the app, and use the camera again, I have the permissions already and the call resolves, showing me the camera.

### Reproduceable Code

```tsx
import React, { useEffect, useState } from 'react';
import { StyleSheet, Text, View } from "react-native";
import { Camera, useCameraDevices, CameraPermissionStatus } from 'react-native-vision-camera';
import ActivityIndicatorText from './ActivityIndicatorText';

const styles = StyleSheet.create({
    container: {
        flex: 1,
        alignItems: 'center',
        justifyContent: 'center'
    }
})

export default function CameraScreen(): JSX.Element {
    const [cameraPermission, setCameraPermission] = useState<CameraPermissionStatus>("not-determined");
    const [error, setError] = useState<string>("")

    useEffect(() => {
        const requestPermissions = async () => {
            console.log("Requesting camera permission");
            const permission = await Camera.requestCameraPermission();
            console.log("Permission given: ", permission);

            if (permission !== "authorized") {
                setError("Not allowed to access camera");
            }

            setCameraPermission(permission);
        }
        
        requestPermissions();
    });

    const devices = useCameraDevices('wide-angle-camera');
    const device = devices.back;

    const renderCamera = () => {
        if (error) {
            return (<Text>Error: {error}</Text>);
        }

        if (cameraPermission === "not-determined") {
            return <ActivityIndicatorText size="large" text="Requesting permission..." />
        }    

        if (device == null) {
            return (<Text>No camera device could be found on your device.</Text>)
        }

        return (
            <Camera
                style={StyleSheet.absoluteFill}
                device={device}
                isActive={true}
            />
        )
    }

    return (
        <View style={styles.container}>
            {renderCamera()}
        </View>
    )
};
```

### What happened instead?

It seems the call `await Camera.requestCameraPermission()` does not resolve when it requests for camera permissions.
I would expect the call to resolve so that I can use the camera.

### Device

Android v11 (Realme 6 Pro)

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ssangram111, 2023-01-24:
const requestCameraPermission = async () => {
    try {
      const granted = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.CAMERA,
        {
          title: 'Cool Photo App Camera Permission',
          message:
            'Cool Photo App needs access to your camera ' 
        },
      );
      if (granted === PermissionsAndroid.RESULTS.GRANTED) {
        console.log('You can use the camera');
      } else {
        console.log('Camera permission denied');
      }
    } catch (err) {
      console.warn(err);
    }
  };

jlemein, 2023-02-03:
I managed to get it working too by using `PermissionsAndroid` as a workaround.
It is still a bug in `react-native-vision-camera` though.

alexhernandez, 2023-02-09:
I was able to get it to work by doing the following:

**Android Manifest:**
```xml
  <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE"/>
  <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"/>
  <uses-permission android:name="android.permission.CAMERA" />
  <uses-permission android:name="android.permission.RECORD_AUDIO" />

  <uses-feature android:name="android.hardware.camera.any" android:required="false" />
  <uses-feature android:name="android.hardware.camera" android:required="false" />

```

**Methods:**
```javascript
async function checkPermissions() {
  let cameraPermission: CameraPermissionStatus = 'not-determined';
  let microphonePermission: CameraPermissionStatus = 'not-determined';

  try {
    cameraPermission = await RNCamera.getCameraPermissionStatus();
    microphonePermission = await RNCamera.getMicrophonePermissionStatus();
  } catch (e) {
    console.log(e, 'checkPermissions');
  }

  return {
    camera: cameraPermission,
    microphone: microphonePermission,
  };
}

async function requestPermissions() {
  let cameraPermission: CameraPermissionStatus = 'denied';
  let microphonePermission: CameraPermissionStatus = 'denied';

  try {
    cameraPermission = await RNCamera.requestCameraPermission();
    microphonePermission = await RNCamera.requestMicrophonePermission();
  } catch (e) {
    console.log(e, 'requestPermissions');
  }

  return {
    camera: cameraPermission,
    microphone: microphonePermission,
  };
}
```

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy