margelo/react-native-vision-camera #1426: 🐛 Getting  A  signal 11 (SIGSEGV), code 2 (SEGV_ACCERR) when camera captures a qr code

yousrasd, 2023-01-24:
### What were you trying to do?

We are using react-native-vision-camera with vision-camera-code-scanner to capture QR code.

We use the useScanBarcodes hook to detect QR codes. When testing our application, the app crashes whenever it tries to capture a QR code. When it captures normal text (containing no QR code), the camera continues working. However, whenever it finds a QR code in its viewport, it abruptly crashes. You can view this behavior bellow:

https://user-images.githubusercontent.com/63681471/214335505-f713cd70-05c8-46f2-bc58-5dfc5ac65fbc.mp4

iPhone devices are working fine. They are able to capture the QR code. 

### Reproduceable Code

```tsx
import React, {useEffect, useRef, useState } from 'react';
import { Linking, StyleSheet, View } from 'react-native';
  import {
  Camera,
 } from 'react-native-vision-camera';
import { BarcodeFormat, useScanBarcodes } from 'vision-camera-code-scanner';

const VisionCamera = ({ onRead, isCameraActive, externalStyle = {} }) => {
  const camera = useRef(null);
  const [backCamera, setBackCamera] = useState(undefined);
  const [qrCode, setQrCode] = useState(null);

  const [frameProcessor, barcodes] = useScanBarcodes([
    BarcodeFormat.QR_CODE,  
  ]);
 

  useEffect(() => {
    const getDevices = async () => {
      try {
        const devices = await Camera.getAvailableCameraDevices();
        setBackCamera(devices[0]);
        return devices;
      } catch (err) {
      }
    };
    getDevices();
  }, []);
 
  const onError =  (error) => {
    if (__DEV__) {
      console.log('Camera run time error', error);
    }
  };

  return (
    backCamera && isCameraActive && (
      <>
        <Camera
          photo={false}
          enableHighQualityPhotos={false}
          ref={camera}
          style={[styles.camera, styles.cameraStyle]}
          isActive={true}
          orientation="portrait"
          device={backCamera}
          frameProcessor={
            backCamera?.supportsParallelVideoProcessing && !qrCode
              ? frameProcessor
              : undefined
          }
          frameProcessorFps={1}
          audio={false}
          onError={onError}
        />
      </>
    )
  );
};
```

### What happened instead?

Whenever the camera detects a QR code, it crashes. I found the following stacktract in Android Studio related to this crash. 

`    A  signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0x78b8f02c18
2023-01-23 18:01:54.123 22757-22757 DEBUG                   crash_dump64                         A      x0  00000078b8f02c18  x1  0000000000000000  x2  b4000079fbfd8770  x3  0000000000000000
2023-01-23 18:01:54.123 22757-22757 DEBUG                   crash_dump64                         A      x4  0000000000000008  x5  0000006f3ad39170  x6  00000000ebad808a  x7  00000000ebad808b
2023-01-23 18:01:54.123 22757-22757 DEBUG                   crash_dump64                         A      x8  434c4e47432b2b00  x9  0000006f3ad38d70  x10 0000006f3ad38d70  x11 0000000000000000
2023-01-23 18:01:54.123 22757-22757 DEBUG                   crash_dump64                         A      x12 0000006f3ad370f8  x13 0000000000000000  x14 0000000000000008  x15 0000000000000008
2023-01-23 18:01:54.124 22757-22757 DEBUG                   crash_dump64                         A      x16 0000000000000020  x17 0000000000000020  x18 0000006f0a7c2000  x19 0000006f3ad38850
2023-01-23 18:01:54.124 22757-22757 DEBUG                   crash_dump64                         A      x20 b4000079fbfd8770  x21 0000000000000002  x22 00000078b8e77048  x23 b40000799c002530
2023-01-23 18:01:54.124 22757-22757 DEBUG                   crash_dump64                         A      x24 00000078ad390260  x25 b40000797c0627d0  x26 0000006f3ad39170  x27 0000000000000000
2023-01-23 18:01:54.124 22757-22757 DEBUG                   crash_dump64                         A      x28 0000006f3ad3d000  x29 0000006f3ad38760
2023-01-23 18:01:54.124 22757-22757 DEBUG                   crash_dump64                         A      lr  00000078cdac8c90  sp  0000006f3ad38760  pc  00000078cdaccbdc  pst 0000000060000000`

### Device

Motorola G Stylus (Android 12)

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

vrakovs-abc, 2023-01-31:
The same, error occurs in this method:
`__scanCodes(frame, types, options);`

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀 

Try upgrading to V3 if you can, it has a much more stable yet flexible foundation.