margelo/react-native-vision-camera #1427: "Expected directory 'node_modules/react-native/android' to contain exactly one file, however, it contains no files"🐛

jason840917gmail, 2023-01-24:
### What were you trying to do?

Just started a new react native project and install all the necessaries dependencies  to access the camera. I had seen that this issue always comes back with every new  version of RN or React native vision camera.
Dependencies: 
```
"@react-navigation/drawer": "^6.5.7",
    "@react-navigation/native": "^6.1.2",
    "latest": "^0.2.0",
    "patch-package": "^6.5.1",
    "react": "18.2.0",
    "react-native": "^0.71.1",
    "react-native-gesture-handler": "^2.9.0",
    "react-native-reanimated": "^2.14.4",
    "react-native-safe-area-context": "^4.5.0",
    "react-native-screens": "^3.19.0",
    "react-native-vision-camera": "^2.15.2"
```

### What happened instead?

[image]

### Device

Android 12

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jason840917gmail, 2023-01-24:
found this on the last release of react native:
Starting from React Native 0.71, we're not shipping the `/android` folder inside the React Native NPM package
anymore due to sizing constraints on NPM. The Android artifacts are distributed via Maven Central.
You can read more about it in this RFC:
https://github.com/react-native-community/discussions-and-proposals/pull/508

If you're a library author and you're manipulating the React Native .aar files, to extract headers,
extract `.so` files or do anything with it, you're probably doing something wrong. React Native
0.71 ships with all the necessary logic to let you consume it transparently by just using:

```
implementation("com.facebook.react:react-android")
// or to keep backward compatibility with older versions of React Native:
implementation("com.facebook.react:react-native:+")

DanielGannage, 2023-01-25:
Can confirm, I get the same error with RN 0.71.1

Currently our project cannot barcode scan, which requires older and deprecated modules that are in need of patching. How long do you think it will take before these artifacts get put in the right spot (i.e. the module gets fixed)?

markjsc, 2023-01-26:
Confirming that we're seeing the same thing with React Native 0.71.1.

Would definitely appreciate any updates or workarounds. We're going to stay on React Native 0.70.x until this is resolved.

Side note - Thanks for all the work on this project!!

zzz08900, 2023-01-26:
Duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/1418

closed 2023-09-01 by mrousavy