margelo/react-native-vision-camera #1431: 🐛 The Android Gradle plugin supports only kotlin-android-extensions Gradle plugin version 1.6.20 and higher

srymarz, 2023-01-25:
### What were you trying to do?

I want build app using react-native-vision-camera: 2.15.2 for react-native 0.70.6

### What happened instead?

* What went wrong:
The Android Gradle plugin supports only kotlin-android-extensions Gradle plugin version 1.6.20 and higher.
The following dependencies do not satisfy the required version:
project ':react-native-actions-shortcuts' -> org.jetbrains.kotlin:kotlin-gradle-plugin:1.6.10

### Device

Android

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

zzz08900, 2023-01-26:
Why do you think react-native-actions-shortcuts has anything to do with react-native-vision-camera?

alois-beto, 2023-01-26:
I added this in `android/build.gradle`:

```diff
buildscript {
    ext {
          [...]
+        kotlinVersion = "1.6.20"
    }
    [...]
}
```

srymarz, 2023-01-26:
Hello, 
I agree this may not be a problem with the react-native-vision-camera, although it appeared after adding it to the project. I modified my `android/build.gradle` with `kotlinVersion="1.6.20"` but the problem still exists. I will contact the developers of the packages that cause the error with this problem

zzz08900, 2023-01-26:
> Hello, I agree this may not be a problem with the react-native-vision-camera, although it appeared after adding it to the project. I modified my `android/build.gradle` with `kotlinVersion="1.6.20"` [...]

To me it looks like you are having kotlin version conflicts in your dependencies, which, in most cases, should be fixed by specifying kotlin version in top-level build.gradle as suggested by @alois-beto.

You may want to look into your deps and check if some of them are not honoring your buildscript configurations and ask the corrosponding devs to fix it. Last I checked, react-native-vision-camera does honor buildscript configurations.

thestretivibity, 2023-01-26:
I had the same problem after I installed the latest version, and here's I solved it

go to ``build.gradle`` as @alois-beto pointed out
``` 
buildscript {
    ext {
          [...]
        kotlinVersion = "1.6.20"
    }
}
```

and then build the project 2 times!

chernandezlt, 2023-01-27:
I had the same error but including kotlinVersion fixed the issue <3

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy