margelo/react-native-vision-camera #1437: 🐛 An unknown camera error occured.

liancp, 2023-01-28:
### What were you trying to do?

An unknown camera error occured.

### Reproduceable Code

```tsx
An unknown camera error occured.
```

### What happened instead?

An unknown camera error occured.

### Relevant log output

```shell
2023-01-28 14:21:42.888  1292-2082  System.err              com.mrs                              W  java.lang.NullPointerException
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at com.mrousavy.camera.utils.CameraCharacteristicsUtilsKt.getDeviceTypes(CameraCharacteristicsUtils.kt:30)
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at com.mrousavy.camera.CameraViewModule$getAvailableCameraDevices$1.invokeSuspend(CameraViewModule.kt:184)
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
2023-01-28 14:21:43.094  1292-4059  unknown:Re...gWebSocket com.mrs                              W  Couldn't connect to "ws://localhost:8081/message?device=rk3399-Android10%20-%2010%20-%20API%2029&app=com.mrs&clientid=DevSupportManagerImpl", will silently retry
2023-01-28 14:21:43.408  1292-1656  ReactNativeJS           com.mrs                              I  FaceIdentification Component | Camera permission status: authorized
2023-01-28 14:21:44.318  1292-1656  ReactNativeJS           com.mrs                              I  Device Utils | 设备型号: rk3399-Android10
2023-01-28 14:21:44.902  1292-1656  ReactNativeJS           com.mrs                              W  Possible Unhandled Promise Rejection (id: 24):
                                                                                                    unknown/unknown: [unknown/unknown] An unknown camera error occured.
                                                                                                    [unknown/unknown]: [unknown/unknown] An unknown camera error occured.
                                                                                                        at _construct (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:20966:39)
                                                                                                        at apply (native)
                                                                                                        at _construct (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:20971:28)
                                                                                                        at Wrapper (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:20933:25)
                                                                                                        at apply (native)
                                                                                                        at _createSuperInternal (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:229753:382)
                                                                                                        at call (native)
                                                                                                        at CameraError (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:229761:26)
                                                                                                        at apply (native)
                                                                                                        at _createSuperInternal (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:229753:382)
                                                                                                        at apply (native)
                                                                                                        at CameraRuntimeError (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:229810:27)
                                                                                                        at tryParseNativeCameraError (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:229834:38)
                                                                                                        at getAvailableCameraDevices$ (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:229625:66)
                                                                                                        at call (native)
                                                                                                        at tryCatch (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:21704:23)
                                                                                                        at anonymous (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:21687:34)
                                                                                                        at anonymous (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:21729:30)
                                                                                                        at call (native)
                                                                                                        at tryCatch (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:21704:23)
                                                                                                        at invoke (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:21735:30)
                                                                                                        at anonymous (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:21742:19)
                                                                                                        at tryCallOne (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:23267:16)
                                                                                                        at anonymous (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:23348:27)
                                                                                                        at apply (native)
                                                                                                        at anonymous (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:26966:26)
                                                                                                        at _callTimer (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:26871:17)
                                                                                                        at _callImmediatesPass (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:26908:17)
                                                                                                        at callImmediates (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:27080:33)
                                                                                                        at __callImmediates (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:2257:35)
[... cut ...]
Android 10

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-02-09:
I'll try something: https://github.com/mrousavy/react-native-vision-camera/pull/1467

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy