margelo/react-native-vision-camera #1440: How to run CoreML Model❓

davidarnaub7, 2023-01-30:
### Question

I am a complete beginner in swift. I am trying to run a model in CoreML.

From the frame I would have to output a MultiArray(1 × 224 × 224 × 3) entry.

Could someone explain me in detail what I would have to do?

The main problem is that I don't know where and how to use the "object" of my model in the callback function of the frame_processor. As well as I don't know how to cast Frame to MLMultiArray.

I have searched thousands of documents on the internet but I can't find anything and I'm going quite blind.

Any help would be appreciated.

### VisionCamera Version

2.15.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

zzz08900, 2023-01-30:
From my understanding you'll need a custom frame processor plugin for that.

https://mrousavy.com/react-native-vision-camera/docs/guides/frame-processors-plugins-overview

zzz08900, 2023-01-30:
If you REALLY want to run AI in the JS world, I recall there's one frame processor plugin that turns frame into base64 string of its content  lying around somewhere.

davidarnaub7, 2023-01-30:
@zzz08900 Sure! I have created the frame processor plugin. The problem I have now is to develop that plugin in Swift.

zzz08900, 2023-01-30:
> @zzz08900 Sure! I have created the frame processor plugin. The problem I have now is to develop that plugin in Swift.

I never tried CoreML myself but with other AI models, what I usually do is take pixel values from imput(in my case it's usually a jpg or png file) and store them into a ndArray, then pass them to the AI model.

Also you probably don't want to pass back the output of your model back to JS.

I guess that's all I can help you with.

nathangabriel27, 2023-02-07:
I have the same problem. I need to implement a corelML model, I can create the plugin in swift, but I don't know how to run the frame in the AI ​​and return the coordinates.

CodingWithNobody, 2023-04-30:
This might help Try this https://www.youtube.com/watch?v=ybiMSEjl5mA

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy