margelo/react-native-vision-camera #1443: 🐛 Execution failed for task ':react-native-vision-camera:extractAARHeaders'.

amiraabdelmalek, 2023-01-31:
### What were you trying to do?

hello, i'am trying to build my react native app for android. and i have this build error. can someone please help me with this.
here's the versions that i'm using.

"react-native": "0.70.6",
"react-native-vision-camera": "^2.15.3",
"react-native-reanimated": "^2.13.0",

my build.gradle:

buildToolsVersion = "32.0.0"
minSdkVersion = 21
compileSdkVersion = 32
targetSdkVersion = 32
googlePlayServicesAuthVersion = "20.2.0"

dependencies {
        classpath("com.android.tools.build:gradle:7.1.1")
        classpath("com.facebook.react:react-native-gradle-plugin")
        classpath("de.undercouch:gradle-download-task:5.0.1")
        classpath 'com.google.gms:google-services:4.3.13'
        classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:1.7.0
    }

### What happened instead?

* What went wrong:
Execution failed for task ':react-native-vision-camera:extractAARHeaders'.
> Could not resolve all files for configuration ':react-native-vision-camera:extractHeaders'.
   > Could not find com.facebook.fbjni:fbjni:headers.
     Searched in the following locations:
       - file:/Users/macos/job-actu-candidate/node_modules/react-native/android/com/facebook/fbjni/fbjni/headers/fbjni-headers.pom
       - file:/Users/macos/job-actu-candidate/node_modules/jsc-android/dist/com/facebook/fbjni/fbjni/headers/fbjni-headers.pom
       - https://repo.maven.apache.org/maven2/com/facebook/fbjni/fbjni/headers/fbjni-headers.pom
       - https://dl.google.com/dl/android/maven2/com/facebook/fbjni/fbjni/headers/fbjni-headers.pom
       - https://www.jitpack.io/com/facebook/fbjni/fbjni/headers/fbjni-headers.pom
       - https://maven.google.com/com/facebook/fbjni/fbjni/headers/fbjni-headers.pom
     Required by:
         project :react-native-vision-camera

### Device

Pixel 3a API 32

### VisionCamera Version

^2.15.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jpabara, 2023-01-31:
Same issue here

yormy, 2023-01-31:
Same here, running on an expo-bare workflow

MahammadNuriyev62, 2023-01-31:
i have exactly the same issue!

kyrsanter, 2023-01-31:
The same 
RN 0.70.1
    "react-native-vision-camera": {
      "version": "2.15.3",

yormy, 2023-01-31:
Partly resolved by following the instructions for : https://docs.expo.dev/versions/latest/sdk/reanimated/
Compiles and builds and app starts, but I have some js errors elsewhere
`
 ERROR  ReferenceError: Can't find variable: __detect
 ERROR  Invariant Violation: Failed to call into JavaScript module method AppRegistry.runApplication()
`

jpabara, 2023-01-31:
@yormy you mean this part?

module.exports = function (api) {
  api.cache(true);
  return {
    presets: ['babel-preset-expo'],
    plugins: ['react-native-reanimated/plugin'],
  };
};

jpabara, 2023-01-31:
I managed to make it work, I know downgrading is not the best idea, but version 2.15.3 of react-native-vision-camera was released yesterday. So I changed package.json to version 2.15.2 and it worked fine after 3 attempts of building the app

yormy, 2023-01-31:
Did you make other changes. I get js errors on the scanner.tsx code

jpabara, 2023-01-31:
Nothing else, I use react-native CLI 

Versions are:
`
"react-native": "0.67.5",
"react": "17.0.2",
"react-native-vision-camera": "2.15.2",
`

SeongwoonHong, 2023-01-31:
i'm running into the same issue.  any luck so far?

MahammadNuriyev62, 2023-01-31:
could only fix by downgrading
`react-native-reanimated` to exactly `2.10.0`
`react-native-vision-camera ` to exactly `2.14.1`

waiting for better solutions from you)

GSFZamai, 2023-01-31:
> I managed to make it work, I know downgrading is not the best idea, but version 2.15.3 of react-native-vision-camera was released yesterday. So I changed package.json to version 2.15.2 and it worked fine after 3 attempts of building the app

TThis worked for me. Updated today and got the this error, before downgrading to 2.15.2, build successfully

jbohman, 2023-01-31:
I have this issue on version 2.15.2. It worked to build locally but in CI it started to fail. After running `./gradlew clean` it started to happen for me locally as well.

SeongwoonHong, 2023-01-31:
`"react-native": "0.66.5"`
`"react-native-vision-camera": "2.14.1"`
`"react-native-reanimated": "^2.10.0"`
This DID NOT work for me.

`"react-native": "0.66.5"`
`"react-native-vision-camera": "2.14.1"`
`"react-native-reanimated": "2.10.0"`
This DID NOT work for me

`"react-native": "0.66.5"`
`"react-native-vision-camera": "2.15.2"`
`"react-native-reanimated": "^2.10.0"`
This DID NOT work for me

Artefans, 2023-01-31:
@mrousavy check pls

mrousavy, 2023-01-31:
> @mrousavy check pls

stop tagging me. I can't reproduce this error, VisionCamera builds in all my apps. If you want to fix this, submit a PR (or pay me to fix it).

marcelmeulemans, 2023-01-31:
Downgrading to 2.15.2 should work but it seems that sometimes/somehow the 2.15.3 build.gradle file is cached somewhere. Check the the last lines match the 2.15.2 version:
https://github.com/mrousavy/react-native-vision-camera/blob/v2.15.2/android/build.gradle#L601
For me a `yarn cache clean` and `yarn add react-native-vision-camera@2.15.2` restored the correct build.gradle.

kartefan, 2023-02-01:
The same
RN: 0.70.6
react-native-vision-camera: 2.15.3

`
Build file '/xxx/node_modules/react-native-vision-camera/android/build.gradle' line: 596

* What went wrong:
Execution failed for task ':react-native-vision-camera:extractAARHeaders'.
> Could not resolve all files for configuration ':react-native-vision-camera:extractHeaders'.
   > Could not find com.facebook.fbjni:fbjni:headers.
     Searched in the following locations:
       - file:/xxx/node_modules/react-native/android/com/facebook/fbjni/fbjni/headers/fbjni-headers.pom
       - file:/xxx/node_modules/jsc-android/dist/com/facebook/fbjni/fbjni/headers/fbjni-headers.pom
       - https://repo.maven.apache.org/maven2/com/facebook/fbjni/fbjni/headers/fbjni-headers.pom
       - https://dl.google.com/dl/android/maven2/com/facebook/fbjni/fbjni/headers/fbjni-headers.pom
       - https://www.jitpack.io/com/facebook/fbjni/fbjni/headers/fbjni-headers.pom
     Required by:
         project :react-native-vision-camera

`

SeongwoonHong, 2023-02-01:
@marcelmeulemans  hmm I'm getting this error
```
> Configure project :react-native-reanimated
Native libs debug enabled: false
Android gradle plugin: 4.2.0
Gradle: 6.9
building Reanimated2

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.9/userguide/command_line_interface.html#sec:command_line_warnings

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':react-native-reanimated'.
> com.android.builder.errors.EvalIssueException: CMake '3.18.1' was not found in SDK, PATH, or by cmake.dir property.

```
Have you come across this before?
`react-native-reanimated: "2.10.0"` and `react-native-vision-camera: "2.15.2"`

ravin-amplework, 2023-02-01:
WORKED FOR ME  >>
 downgrad to   2.15.1

amiraabdelmalek, 2023-02-01:
> Did you make other changes. I get js errors on the scanner.tsx code Do you use react-native setup or expo bare setup ?

No I use react native cli setup

zzz08900, 2023-02-01:
So it seems 2.15.3 breaks on RN < 0.71?
I'll give a try and see what's going on

But for some of the problems in this thread, like CMake '3.18.1' was not found, probably has nothing to do with react-native-vision-camera.

---------------------------------

OK I think I see the problem.
Version 2.15.3 does break on RN < 0.71

So here's the plan:
Plan A: you can downgrade react-native-vision-camera to 2.15.2.
Plan B: you edit `react-native-vision-camera/android/build.gradle`, search for `extractHeaders("com.facebook.fbjni:fbjni:headers:+")` and replace it with `extractHeaders("com.facebook.fbjni:fbjni:+:headers")`

It's probably a typo when they made the fix for RN 0.71

amiraabdelmalek, 2023-02-01:
Downgrading to 2.15.2 worked for me.
"react-native-reanimated": "^2.13.0",
"react-native-vision-camera": "2.15.2",
 "react-native": "0.70.6",
"react": "^18.2.0",
Working with react native cli.

ChristophGr, 2023-02-01:
Yes. It happens in projects using rn < 0.71. but @zzz08900 fix is exactly what's wrong:
https://github.com/mrousavy/react-native-vision-camera/pull/1447

mrousavy, 2023-02-01:
Hey @amiraabdelmalek, I just released 2.15.4 which includes the fix from #1447, can you test if that works for you? :)

SeongwoonHong, 2023-02-02:
I was able to resolve this after downgrading the version to 2.15.2 and wiping data on emulator

ogard, 2023-02-02:
I can confirm that fix from #1447 solves the issue.
```
"react-native": "0.70.6"
"react-native-vision-camera": "^2.15.4"
```

krisidmisso, 2023-04-20:
I have tried all the downgrades mentioned above and for me it doesnt work. I have the same issue:

```
FAILURE: Build failed with an exception.

* Where:
Build file '/.../node_modules/react-native-vision-camera/android/build.gradle' line: 596

* What went wrong:
Execution failed for task ':react-native-vision-camera:extractAARHeaders'.
> Could not resolve all files for configuration ':react-native-vision-camera:extractHeaders'.
   > Could not resolve com.facebook.fbjni:fbjni:+.
     Required by:
         project :react-native-vision-camera
      > Failed to list versions for com.facebook.fbjni:fbjni.
         > Unable to load Maven meta-data from https://repo.maven.apache.org/maven2/com/facebook/fbjni/fbjni/maven-metadata.xml.
            > Could not HEAD 'https://repo.maven.apache.org/maven2/com/facebook/fbjni/fbjni/maven-metadata.xml'.
               > repo.maven.apache.org: nodename nor servname provided, or not known
      > Failed to list versions for com.facebook.fbjni:fbjni.
         > Unable to load Maven meta-data from https://repo.maven.apache.org/maven2/com/facebook/fbjni/fbjni/maven-metadata.xml.
            > Could not HEAD 'https://repo.maven.apache.org/maven2/com/facebook/fbjni/fbjni/maven-metadata.xml'.
               > repo.maven.apache.org

```

Also #1447 is already applied since I have the latest version 2.15.4. It was working 1 week ago. Now that I tried with `./gradlew clean` it doesn't work anymore.

@mrousavy Can it be related due to a change done in the aar file? Since it is always re-downloaded whenever I try to build for android

krisidmisso, 2023-04-21:
For anyone getting stuck with this issue I got it working with the following steps:

1. clear any cache of gradle and npm/yarn
2. delete .lock files (package-json.lock, yarn.lock)
3. restart pc without doing anything else (dont laugh, it works some times when you are desperate)
4. npm install/yarn install
5. update `android/build.gradle` build tools versions (from 31 to 32):
```
buildscript {
    ext {
        buildToolsVersion = "32.0.0"
        minSdkVersion = 21
        compileSdkVersion = 32
        targetSdkVersion = 32

        VisionCameraCodeScanner_targetSdkVersion = 32
        VisionCameraCodeScanner_compileSdkVersion = 32
       
 ```
6. `./gradlew clean` inside android folder
7. npm run android

jg-stack, 2023-04-28:
Was facing the same issue with this versioning:

` "react-native": "0.71.7",
"react-native-vision-camera": "2.15.4"`

Solved the issue for me by updating `react-native-reanimated` to `v2.17.0` (the last version before they moved to v3) then doing a full `react-native-clean-project`

pavelmuratev, 2023-07-25:
What fixed the issue for us was: 
`./gradlew clean && ./gradlew --refresh-dependencies`

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

Soul-Clinic, 2023-08-26:
code ../react-native-project/node_modules/react-native-vision-camera/android/build.gradle
Remove or comment:
    
    apply plugin: 'kotlin-android-extensions'

closed 2023-09-01 by mrousavy

prantaDutta, 2023-10-12:
> What fixed the issue for us was: `./gradlew clean && ./gradlew --refresh-dependencies`

This works

JocheAttah, 2023-10-17:
> What fixed the issue for us was: `./gradlew clean && ./gradlew --refresh-dependencies`

This worked for me too

aakarshan-raj, 2024-01-16:
Tired of React native ecosystem  🙂