margelo/react-native-vision-camera #1445: 🐛 Camera orientation problem

galkinsden, 2023-01-31:
### What were you trying to do?

If all orientations are available for the device, the camera rotates with the device.

### Reproduceable Code

```tsx
import React, { memo, useEffect } from 'react';
import { StyleSheet } from 'react-native';
import Orientation from 'react-native-orientation-locker';
import { Camera as CameraVision, useCameraDevices } from 'react-native-vision-camera';

const Camera = () => {
  const devices = useCameraDevices();
  const device = devices?.back;

  useEffect(() => {
    Orientation.unlockAllOrientations();
    return () => {
      Orientation.lockToPortrait();
    };
  }, []);

  return <CameraVision style={StyleSheet.absoluteFill} device={device} isActive photo enableZoomGesture />;
};

export default memo(Camera);
```

### What happened instead?

If all orientations are available for the device, the camera rotates with the device. The problem is reproduced only on version >= 2.15.3 

https://user-images.githubusercontent.com/46136101/215867915-209860b5-a166-4d44-9ee4-635a5d43c464.MP4

### Device

iPhone SE2

### VisionCamera Version

2.15.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

alexhernandez, 2023-02-09:
```javascript
<CameraVision 
  style={StyleSheet.absoluteFill} 
  device={device} 
  isActive photo 
  orientation="portrait" <---- this fixed it for me
  enableZoomGesture />;
  ```

thorn1997, 2023-02-15:
I have the same issue and still could not fix it, even with the suggested solution above

bqrichards, 2023-02-18:
Same issue.

UmerWarraich95, 2023-02-21:
@galkinsden  I have the same issue for IOS when I change camera rotation. For Android, the rotation is working fine.
@mrousavy  can you look into this issue? I know you worked on this library in your free time.

mrousavy, 2023-02-21:
> @mrousavy can you look into this issue? I know you worked on this library in your free time.

Sorry no I have other things to do right now. If you want this fixed, reach out to us for consulting services: hello@margelo.io

sekcan, 2023-03-03:
Same issue

lucksp, 2023-04-21:
I am not sure I understand the `orientation` prop expected outcome.  If I read [the docs](https://www.react-native-vision-camera.com/docs/api/interfaces/CameraProps#orientation), it appears that it impacts the captured "output".  

By having the following snippet:
```
<Camera
  photo
  orientation={'portrait'}
  device={device}
  isActive={isActive}
  format={formats?.[0]}
 />
```

I understand the docs to say:  _"If I have my device in landscape, the result will be a landscape picture rotated 90 degrees so that it's visible in `portrait` orientation.  Therefore I don't have to turn my head or device to see the photo since I am holding the phone in portrait mode after the picture is taken"_

However, the result is not this: 
On the web previewing my image is not `portrait`
[image]

and neither is it on my app when previewing image:
[image]

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

mrousavy, 2023-09-30:
Hey - I'm tracking Orientation in this feature request/issue now: https://github.com/mrousavy/react-native-vision-camera/issues/1891 ✨ 

Make sure to upvote or sponsor to support this feature, and leave a comment if you have any additional thoughts/ideas.

luisreiskeys, 2024-05-31:
I Just made a module to handle the device orientation
https://www.npmjs.com/package/rn-device-orientation
the usage is pretty simple, and this module get de device orientation, not the interface orientation, so even if the interface is locked on portrait, it will work

to solve on vision camera a jus use like this

[image]

medsabbar, 2025-03-04:
this is the workaround that worked for me

`const getCorrectRotation = async (
  imagePath: string,
  orientation: Orientation,
  cameraPosition: CameraPosition,
) => {
  const exifData = await Exif.getExif(imagePath);
  console.log("exifData", exifData);
  const exifOrientation = exifData?.Orientation || 1;

  // Handle iOS front camera mirroring
  const isFrontCamera = cameraPosition === "front";
  const shouldMirror = isFrontCamera && Platform.OS === "ios";

  // Convert EXIF orientation to rotation degrees
  const rotationMap: { [key: number]: number } = {
    1: 0, // Normal
    2: 0, // Normal
    3: 180, // 180° rotated
    4: 180, // 180° rotated
    5: 90, // 90° rotated (right)
    6: 90, // 90° rotated (right)
    7: -90, // 270° rotated (left)
    8: -90, // 270° rotated (left)
  };

  return {
    rotation: rotationMap[exifOrientation] || 0,
    shouldMirror,
  };
};`

`const { rotation } = await getCorrectRotation(
          image.path,
          image.orientation,
          selectedCamera,
        );
        image = await ImageResizer.createResizedImage(
          image.path,
          image.width,
          image.height,
          "JPEG",
          100,
          rotation,
        );`

PR #1544 (2023-09-30): fix: updated orientation calculations
## What

This PR should fix the issues in video orientation (preview and recording) being "fixed" to portrait orientation regardless of device orientation in iOS.

Adapted from https://stackoverflow.com/a/36575423

## Changes

* Changed the CameraView orientation extension so that it takes the UIDevice's reported orientation if the userOrientation is not provided
* On rotation change it will update the preview layer orientation and update the preview layer bounds accordingly

## Tested on

* iPhone 14 Pro
* iPad Mini

## Related issues

* Fixes #1539 
* Fi [... cut]