margelo/react-native-vision-camera #1448: ✨ Support for changing the border around Camera.

TideOPL, 2023-02-01:
### What feature or enhancement are you suggesting?

The Camera component currently has no way to add a border and due to this cannot round the corners of the component.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

There are no alternatives for this feature.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

vexogon, 2023-02-03:
I'm having the same issue

severinferard, 2023-02-08:
One alternative would be to use [@react-native-masked-view/masked-view
](https://github.com/react-native-masked-view/masked-view) and to apply the `borderRadius` to the `MaskedView`.

Here is an example:
```tsx
``

severinferard, 2023-02-08:
One alternative would be to use [@react-native-masked-view/masked-view
](https://github.com/react-native-masked-view/masked-view) and to apply the `borderRadius` to the `MaskedView`.

Here is an example:
```tsx
import MaskedView from '@react-native-masked-view/masked-view';

// [...]

<SafeAreaView style={styles.container}>
  <MaskedView
    style={styles.maskedView}
    maskElement={
      <View style={styles.maskWrapper}>
        <View style={styles.mask} />
      </View>
    }
  >
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
    />
  </MaskedView>
</SafeAreaView>

// [...]

const styles = StyleSheet.create({
  container: {
    flex: 1,
    backgroundColor: '#EEE',
  },
  maskedView: {
    flex: 1,
    flexDirection: 'row',
    height: '100%',
  },
  maskWrapper: {
    backgroundColor: 'transparent',
    flex: 1,
    justifyContent: 'center',
    alignItems: 'center',
  },
  mask: {
    aspectRatio: 1,
    width: '80%',
    backgroundColor: 'black',
    borderRadius: 20,
  },
});
```

Which gives:

[image]

vanenshi, 2023-02-24:
I also need this feature

vanenshi, 2023-02-25:
You can also use something like this

```tsx
<View
  style={{
    height: 100,
    borderRadius: 24,
    overflow: "hidden",
  }}
>
  <Camera
    device={device}
    isActive
    style={{
      height: 100,
    }}
  />
</View>;
```

The trick is to use `overflow = "hidden"`

TideOPL, 2023-02-25:
Thank you for this @vanenshi seemed a little silly to use a whole new package just to put a rounded corner on the camera. 👍

vanenshi, 2023-02-25:
No problem, I also didn't want to add a new package just for this feature @TideOPL

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy