margelo/react-native-vision-camera #1449: 🐛 The Android Gradle plugin supports only kotlin-android-extensions Gradle plugin version 1.6.20 and higher.

PortalDK, 2023-02-01:
### What were you trying to do?

react-native run-android

### What happened instead?

The Android Gradle plugin supports only kotlin-android-extensions Gradle plugin version 1.6.20 and higher.
The following dependencies do not satisfy the required version:
project ':react-native-vision-camera' -> org.jetbrains.kotlin:kotlin-gradle-plugin:1.5.30

### Device

emulator

### VisionCamera Version

^2.15.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

celciusjj, 2023-02-01:
same issue here

PortalDK, 2023-02-02:
"react-native": "0.71.1",

Timson020, 2023-02-03:
same issue here

dunapanta, 2023-02-06:
I have the same issue using 
"react-native-vision-camera": "^2.15.4"
"react-native": "0.71.1"

any idea how to solve it?

dunapanta, 2023-02-06:
I managed to make it work with @dcangulo solution, on path
node_modules/react-native-vision-camera/android/gradle.properties
change `VisionCamera_kotlinVersion=1.5.30` to `VisionCamera_kotlinVersion=1.6.20`

but I don't know why this change was not included on version 2.15.4, probably need to test this solution on previous versions of react-native?

edwin5050, 2023-02-06:
This works for me.

fontesrp, 2023-02-06:
The solution described [here](https://github.com/mrousavy/react-native-vision-camera/issues/1352#issuecomment-1346443556) avoids changing lib files. Both work for me. Thanks!

aniketjha9179, 2023-03-14:
Yes its worked for me after updating node_modules/react-native-vision-camera/android/gradle.properties
 change VisionCamera_kotlinVersion=1.5.30 to VisionCamera_kotlinVersion=1.6.20 and after that I got  a dialouge of install Android  SDK build-tools 33 after installing them its worked

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

fontesrp, 2023-08-18:
@mrousavy, thanks for the PR! Very impressive work with Camera2. I'll try it again shortly, but, having recently tried v3 on our app, it seems that as long as [`kotlin-android-extensions`](https://github.com/mrousavy/react-native-vision-camera/pull/1674/files#diff-197b190e4a3512994d2cebed8aff5479ff88e136b8cc7a4b148ec9c3945bd65aR17) is present, the project won't compile with `kotlinVersion` set to `1.8` or higher. I'll try with the lib pointing to the PR though and post here what I find 👍

mrousavy, 2023-08-18:
Ah sorry maybe I can remove that dependency? Not sure if I need it 🤔

closed 2023-09-01 by mrousavy