margelo/react-native-vision-camera #1450: 🐛

hoseinfo, 2023-02-02:
### What were you trying to do?

Try to build iOS project with Xcode.

### What happened instead?

I faced this issue:

### **_'jsi/JSCRuntime.h' file not found_**

### Device

iPhone14(iOS 16.2)

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

hoseinfo, 2023-02-02:
"react-native": "0.71.1"

FDiskas, 2023-02-02:
```patch
diff --git a/node_modules/react-native-vision-camera/ios/React Utils/MakeJSIRuntime.h b/node_modules/react-native-vision-camera/ios/React Utils/MakeJSIRuntime.h
index a428c01..663ee4b 100644
--- a/node_modules/react-native-vision-camera/ios/React Utils/MakeJSIRuntime.h	
+++ b/node_modules/react-native-vision-camera/ios/React Utils/MakeJSIRuntime.h	
@@ -22,7 +22,7 @@
   #include <v8runtime/V8RuntimeFactory.h>
 #else
   // JSC
-  #include <jsi/JSCRuntime.h>
+  #include <React_jsc/JSCRuntime.h>
 #endif
 
 using namespace facebook;
```

zzz08900, 2023-02-03:
I guess a proper title should be: Vision camera breaks on RN 0.71 with Hermes disabled.

(If you enabled Hermes the build script won't be making JSI runtime)

mikebouwmans, 2023-02-06:
> 

Your solution would trigger `'React_jsc/JSCRuntime.h' file not found`. It should be `#include <React_jsc/JSCRuntime.h>`

mikebouwmans, 2023-02-06:
> If you are not using hermes then it could help
> 
> ```diff
> diff --git a/node_modules/react-native-vision-camera/ios/React Utils/MakeJSIRuntime.h b/node_modules/react-native-vision-camera/ios/React Utils/MakeJSIRuntime.h
> index a428c01..663ee4b 100644
> --- a/node_modules/react-native-vision-camera/ios/React Utils/MakeJSIRuntime.h	
> +++ b/node_modules/react-native-vision-camera/ios/React Utils/MakeJSIRuntime.h	
> @@ -22,7 +22,7 @@
>    #include <v8runtime/V8RuntimeFactory.h>
>  #else
>    // JSC
> -  #include <jsi/JSCRuntime.h>
> +  #include <React_jsc/JSCRuntime.h>
>  #endif
>  
>  using namespace facebook;
> ```

This solution would trigger 'React_jsc/JSCRuntime.h' file not found. It should be #include <React_jsc/JSCRuntime.h>

mikebouwmans, 2023-02-06:
Submitted a PR (#1456) with an fix.

closed 2023-02-13 by hoseinfo

AntonioLi1, 2023-02-14:
> If you are not using hermes then it could help
>
> ```diff
> diff --git a/node_modules/react-native-vision-camera/ios/React Utils/MakeJSIRuntime.h b/node_modules/react-native-vision-camera/ios/React [...]

in which file are you adding this?

FDiskas, 2023-02-14:
We are using https://www.npmjs.com/package/patch-package

hzcfy, 2023-02-27:
> > If you are not using hermes then it could help
> > ```diff
> > diff --git a/node_modules/react-native-vision-camera/ios/React Utils/MakeJSIRuntime.h b/node_modules/react-native-vision-camera/ios/React Utils/MakeJSIRuntime.h
> > index a428c01..663ee4b 100644
> > --- a/node_modules/react-native-vision-camera/ios/React Utils/MakeJSIRuntime.h	
> > +++ b/node_modules/react-native-vision-camera/ios/React Utils/MakeJSIRuntime.h	
> > @@ -22,7 +22,7 @@
> >    #include <v8runtime/V8RuntimeFactory.h>
> >  #else
> >    // JSC
> > -  #include <jsi/JSCRuntime.h>
> > +  #include <React_jsc/JSCRuntime.h>
> >  #endif
> >  
> >  using namespace facebook;
> > ```
> 
> This solution would trigger `React_jsc/JSCRuntime.h file not found`. It should be `React-jsc/JSCRuntime.h`

thank you

maskowe, 2023-03-18:
Is there a specific android file at react-native-node_modules/react-native-vision-camera/android that can be changed with patch-package for the same fix?   I'm using "react-native-vision-camera": "2.15.3" and "react-native": "0.71.3" and have hermes disabled.
Thanks for any suggestions.

hzcfy, 2023-03-18:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

maskowe, 2023-03-18:
好，谢谢

maskowe, 2023-03-18:
Nevermind, found it at:
node_modules\react-native-vision-camera\android\src\main\cpp\MakeJSIRuntime.h

Jackwew, 2023-06-28:
> "react-native": "0.71.1"

> thank you

I did it like you did, but it didn't work

m1771vw, 2023-08-30:
I spent two days trying to figure out what was going on, didn't know it was a hermes thing! Thank you!

Changing this in `/node_modules/react-native-vision-camera/ios/React Utils/MakeJSIRuntime.h` worked
```
-  #include <jsi/JSCRuntime.h>
+  #include <React-jsc/JSCRuntime.h>
```

josh-thompson13, 2023-10-26:
@m1771vw I got the exact same "file not found" error. Only thing that makes it work is removing the line completely.

SkioFox, 2024-03-19:
https://github.com/margelo/react-native-worklets-core/issues/150 @hzcfy could you help me any idea? above link has the detail info about the issue in similar

zpinto123, 2024-05-02:
This issue is now present on version 4.0.0/4.0.1.