margelo/react-native-vision-camera #1454: 🐛 Android & RN 0.71: Error at CMakeLists.txt:19 (find_package):   Could not find a package configuration file provided by "fbjni" with any of   the following names:      fbjniConfig.cmake     fbjni-config.cmake

anagovitsyn, 2023-02-03:
### What were you trying to do?

I wanted to use react-native-vision-camera using react-native 0.71 (after upgrading from 0.68)

### What happened instead?

Error while building Vision Camera referencing /....../node_modules/react-native-vision-camera/android
CMake Error at CMakeLists.txt:19
saying:
CMake Error at CMakeLists.txt:19 (find_package):
  Could not find a package configuration file provided by "fbjni" with any of
  the following names:

    fbjniConfig.cmake
    fbjni-config.cmake

  Add the installation prefix of "fbjni" to CMAKE_PREFIX_PATH or set
  "fbjni_DIR" to a directory containing one of the above files.  If "fbjni"
  provides a separate development package or SDK, be sure it has been
  installed.

### Relevant log output

```shell
[CXX1429] error when building with cmake using /Users/asnagovits/Downloads/trash/ProjectName71/node_modules/react-native-vision-camera/android/CMakeLists.txt: -- Android: Targeting API '21' with architecture 'x86', ABI 'x86', and processor 'i686'
-- Android: Selected unified Clang toolchain
-- The C compiler identification is Clang 14.0.1
-- The CXX compiler identification is Clang 14.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/asnagovits/Library/Android/sdk/ndk/24.0.8215888/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/asnagovits/Library/Android/sdk/ndk/24.0.8215888/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring incomplete, errors occurred!
See also "/Users/asnagovits/Downloads/trash/ProjectName71/node_modules/react-native-vision-camera/android/.cxx/Debug/h4m5t754/x86/CMakeFiles/CMakeOutput.log".

C++ build system [configure] failed while executing:
    /Users/asnagovits/Library/Android/sdk/cmake/3.22.1/bin/cmake \
      -H/Users/asnagovits/Downloads/trash/ProjectName71/node_modules/react-native-vision-camera/android \
      -DCMAKE_SYSTEM_NAME=Android \
      -DCMAKE_EXPORT_COMPILE_COMMANDS=ON \
      -DCMAKE_SYSTEM_VERSION=21 \
      -DANDROID_PLATFORM=android-21 \
      -DANDROID_ABI=x86 \
      -DCMAKE_ANDROID_ARCH_ABI=x86 \
      -DANDROID_NDK=/Users/asnagovits/Library/Android/sdk/ndk/24.0.8215888 \
      -DCMAKE_ANDROID_NDK=/Users/asnagovits/Library/Android/sdk/ndk/24.0.8215888 \
      -DCMAKE_TOOLCHAIN_FILE=/Users/asnagovits/Library/Android/sdk/ndk/24.0.8215888/build/cmake/android.toolchain.cmake \
      -DCMAKE_MAKE_PROGRAM=/Users/asnagovits/Library/Android/sdk/cmake/3.22.1/bin/ninja \
      "-DCMAKE_CXX_FLAGS=-fexceptions -frtti -std=c++1y -DONANDROID" \
      -DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/Users/asnagovits/Downloads/trash/ProjectName71/node_modules/react-native-vision-camera/android/build/intermediates/cxx/Debug/h4m5t754/obj/x86 \
      -DCMAKE_RUNTIME_OUTPUT_DIRECTORY=/Users/asnagovits/Downloads/trash/ProjectName71/node_modules/react-native-vision-camera/android/build/intermediates/cxx/Debug/h4m5t754/obj/x86 \
      -DCMAKE_BUILD_TYPE=Debug \
      -DCMAKE_FIND_ROOT_PATH=/Users/asnagovits/Downloads/trash/ProjectName71/node_modules/react-native-vision-camera/android/.cxx/Debug/h4m5t754/prefab/x86/prefab \
      -B/Users/asnagovits/Downloads/trash/ProjectName71/node_modules/react-native-vision-camera/android/.cxx/Debug/h4m5t754/x86 \
      -GNinja \
      -DANDROID_STL=c++_shared \
      -DREACT_NATIVE_VERSION=71 \
      -DNODE_MODULES_DIR=/Users/asnagovits/Downloads/trash/ProjectName71/node_modules \
      -DFOR_HERMES=true \
      -DJS_RUNTIME_DIR=/Users/asnagovits/Downloads/trash/ProjectName71/node_modules/react-native/sdks/hermes
  from /Users/asnagovits/Downloads/trash/ProjectName71/node_modules/react-native-vision-camera/android
CMake Error at CMakeLists.txt:19 (find_package):
  Could not find a package configuration file provided by "fbjni" with any of
  the following names:

    fbjniConfig.cmake
    fbjni-config.cmake

  Add the installation prefix of "fbjni" to CMAKE_PREFIX_PATH or set
  "fbjni_DIR" to a directory containing one of the above files.  If "fbjni"
  provides a separate development package or SDK, be sure it has been
  installed.
Unresolved module dependency :react-native-reanimated (:) in :vision-camera-code-scanner (BuildId(buildRoot=/Users/asnagovits/Downloads/trash/ProjectName71/android)). Neither the source set nor the artifact property was populated by the Android Gradle plugin.
Unresolved module dependency :react-native-reanimated (:) in :vision-camera-code-scanner (BuildId(buildRoot=/Users/asnagovits/Downloads/trash/ProjectName71/android)). Neither the source set nor the artifact property was populated by the Android Gradle plugin.
Unresolved module dependency :react-native-reanimated (:) in :vision-camera-code-scanner (BuildId(buildRoot=/Users/asnagovits/Downloads/trash/ProjectName71/android)). Neither the source set nor the artifact property was populated by the Android Gradle plugin.
```

### Device

Android Studio, Mac M1 Pro

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

anagovitsyn, 2023-02-03:
**project build.gradle:**
import org.apache.tools.ant.taskdefs.condition.Os

// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    ext {
        kotlinVersion = '1.5.30'
        buildToolsVersion = "29.0.2"
        buildToolsVersion = "31.0.0"
        minSdkVersion = 24
        compileSdkVersion = 31
        targetSdkVersion = 31
        supportLibVersion = "29.0.2"
        androidXCore = "1.0.2" // <-- Add this. Check versions here: https://developer.android.com/jetpack/androidx/releases/core

        if (System.properties['os.arch'] == "aarch64") {
            // For M1 Users we need to use the NDK 24 which added support for aarch64
            ndkVersion = "24.0.8215888"
        } else {
            // Otherwise we default to the side-by-side NDK version from AGP.
            ndkVersion = "21.4.7075529"
        }
    }
    repositories {
        mavenCentral()
        jcenter()
        google()
    }
    dependencies {
        classpath('com.android.tools.build:gradle:7.4.0')
        classpath("com.facebook.react:react-native-gradle-plugin")
        classpath("de.undercouch:gradle-download-task:4.1.2")
        classpath 'com.google.gms:google-services:4.0.1'
        classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:$kotlinVersion"
        classpath "org.jetbrains.kotlin:kotlin-android-extensions:$kotlinVersion"
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        exclusiveContent {
            // We get React Native's Android binaries exclusively through npm,
            // from a local Maven repo inside node_modules/react-native/.
            // (The use of exclusiveContent prevents looking elsewhere like Maven Central
            // and potentially getting a wrong version.)
            filter {
                includeGroup "com.facebook.react"
            }
            forRepository {
                maven {
                    // NOTE: if you are in a monorepo, you may have "$rootDir/../../../node_modules/react-native/android"
                    url "$rootDir/../node_modules/react-native/android"
                }
            }
        }

        google()
        mavenLocal()

        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url("$rootDir/../node_modules/react-native/android")
        }
        maven {
            // Android JSC is installed from npm
            url("$rootDir/../node_modules/jsc-android/dist")
        }
        maven { url 'https://developer.huawei.com/repo/' }
        mavenCentral {
            // We don't want to fetch react-native from Maven Central as there are
            // older versions over there.
            content {
                excludeGroup "com.facebook.react"
            }
        }

        mavenCentral()
        jcenter()
        maven { url 'https://www.jitpack.io' }
    }
}

anagovitsyn, 2023-02-03:
**app build.gradle:**
```
apply plugin: "com.android.application"
apply plugin: "com.facebook.react"
apply from: "../../node_modules/react-native-vector-icons/fonts.gradle"
apply plugin: 'kotlin-android'
apply plugin: 'kotlin-android-extensions'

import com.android.build.OutputFile

/**
 * This is the configuration block to customize your React Native Android app.
 * By default you don't need to apply any configuration, just uncomment the lines you need.
 */
react {
    /* Folders */
    //   The root of your project, i.e. where "package.json" lives. Default is '..'
    // root = file("../")
    //   The folder where the react-native NPM package is. Default is ../node_modules/react-native
    // reactNativeDir = file("../node_modules/react-native")
    //   The folder where the react-native Codegen package is. Default is ../node_modules/react-native-codegen
    // codegenDir = file("../node_modules/react-native-codegen")
    //   The cli.js file which is the React Native CLI entrypoint. Default is ../node_modules/react-native/cli.js
    // cliFile = file("../node_modules/react-native/cli.js")

    /* Variants */
    //   The list of variants to that are debuggable. For those we're going to
    //   skip the bundling of the JS bundle and the assets. By default is just 'debug'.
    //   If you add flavors like lite, prod, etc. you'll have to list your debuggableVariants.
    // debuggableVariants = ["liteDebug", "prodDebug"]

    /* Bundling */
    //   A list containing the node command and its flags. Default is just 'node'.
    // nodeExecutableAndArgs = ["node"]
    //
    //   The command to run when bundling. By default is 'bundle'
    // bundleCommand = "ram-bundle"
    //
    //   The path to the CLI configuration file. Default is empty.
    // bundleConfig = file(../rn-cli.config.js)
    //
    //   The name of the generated asset file containing your JS bundle
    // bundleAssetName = "MyApplication.android.bundle"
    //
    //   The entry file for bundle generation. Default is 'index.android.js' or 'index.js'
    // entryFile = file("../js/MyApplication.android.js")
    //
    //   A list of extra flags to pass to the 'bundle' commands.
    //   See https://github.com/react-native-community/cli/blob/main/docs/commands.md#bundle
    // extraPackagerArgs = []

    /* Hermes Commands */
    //   The hermes compiler command to run. By default it is 'hermesc'
    // hermesCommand = "$rootDir/my-custom-hermesc/bin/hermesc"
    //
    //   The list of flags to pass to the Hermes compiler. By default is "-O", "-output-source-map"
    // hermesFlags = ["-O", "-output-source-map"]
}

/**
 * Set this to true to create four separate APKs instead of one,
 * one for each native architecture. This is useful if you don't
 * use App Bundles (https://developer.android.com/guide/app-bundle/)
 * and want to have separate APKs to upload to the Play Store.
 */
def enableSeparateBuildPerCPUArchitecture = false

/**
 * Set this to true to Run Proguard on Release builds to minify the Java bytecode.
 */
[... 213 lines cut ...]
    }

    debugImplementation("com.facebook.flipper:flipper-fresco-plugin:${FLIPPER_VERSION}")
    if (hermesEnabled.toBoolean()) {
        implementation("com.facebook.react:hermes-android")
    } else {
        implementation jscFlavor
    }
}

// Run this once to be able to run the application with BUCK
// puts all compile dependencies into folder libs for BUCK to use
//task copyDownloadableDepsToLibs(type: Copy) {
//    from configurations.compile
//    into 'libs'
//}

apply plugin: 'com.google.gms.google-services'
googleServices {
    disableVersionCheck = true
}

apply from: file("../../node_modules/@react-native-community/cli-platform-android/native_modules.gradle"); applyNativeModulesAppBuildGradle(project)

```

vatsal-gadhiya-searce, 2023-02-06:
Same issue. Any update here?

realsteel85, 2023-07-04:
Same issue :(

Isihaaaa, 2023-07-21:
SAME FUCKING ISSUE 
:( I love it

anagovitsyn, 2023-07-22:
I have finally passed this issue through. Steps to solve:

1. kotlinVersion = '1.6.20' in android/build.gradle

2 comment out com.facebook.react in exclusiveContent  in android/build.gradle

`allprojects {
    repositories {
//        exclusiveContent {
//            // We get React Native's Android binaries exclusively through npm,
//            // from a local Maven repo inside node_modules/react-native/.
//            // (The use of exclusiveContent prevents looking elsewhere like Maven Central
//            // and potentially getting a wrong version.)
//            filter {
//                includeGroup "com.facebook.react"
//            }
//            forRepository {
//                maven {
//                    // NOTE: if you are in a monorepo, you may have "$rootDir/../../../node_modules/react-native/android"
//                    url "$rootDir/../node_modules/react-native/android"
//                }
//            }
//        }
`

closed 2023-07-22 by anagovitsyn

react06, 2023-09-06:
Any Update on this Issue? Any help will be appreciated. Thank you in Advance.

fernandopascoalbr, 2024-07-18:
> I have finally passed this issue through. Steps to solve:
> 
> 1. Set kotlinVersion = '1.6.20' in android/build.gradle
> 2. Comment out com.facebook.react in exclusiveContent  in android/build.gradle
> 
> // exclusiveContent { // // We get React Native's Android binaries exclusively through npm, // // from a local Maven repo inside node_modules/react-native/. // // (The use of exclusiveContent prevents looking elsewhere like Maven Central // // and potentially getting a wrong version.) // filter { // includeGroup "com.facebook.react" // } // forRepository { // maven { // // NOTE: if you are in a monorepo, you may have "$rootDir/../../../node_modules/react-native/android" // url "$rootDir/../node_modules/react-native/android" // } // } // }

Thanks @anagovitsyn , i comment my "exclusiveContent". It`s work for me.